dfki-ric/phobos #10: Update modeling tutorial

kavonszadkowski, 2015-02-17:
Related to #5, we need to update the modeling tutorial to include all changes since the last release. 

We should also seriously get round to getting our hands dirty on videos (#3), though this is something we could add after the release.

kavonszadkowski, 2015-02-27:
Are you planning to work on this today, @SamuelYaron? I'd like to make a few changes to various parts of the documentation and I'd like to avoid merge conflicts.

SamuelYaron, 2015-02-27:
In order of my discomforts I will work on it later this day. So there will be no merge conflicts when you edit the documentation in the next hours

kavonszadkowski, 2015-02-27:
Overhauling the wiki is done. I updated everything and added a few missing hints I could think of. It would be great if you could still add the description of how to add the sensors.

closed 2015-03-09 by SamuelYaron