dfki-ric/phobos #5: Update example model

kavonszadkowski, 2015-02-17:
There have been quite a few changes in how data is handled since the last release, as well as additional features such as sensors. Thus for 0.6, we need to update our example robot model in order for it to serve as a helpful "cheat sheet" for people.

traversaro, 2015-02-17:
Hi, congratulations for the great software. 
I read here that you plan to add sensors information to your model editor. Which kind of format you plan to use for encoding sensors information in the URDF ?

In the `urdfdom` repository there is an open discussion [1] for new URDF specifications regarding sensors. Do you think it could be worth to share your proposal for simplify integration and interoperability of different software using sensor information from URDF?  

[1] : https://github.com/ros/urdfdom/pull/28

kavonszadkowski, 2015-02-17:
Thanks a lot for the very relevant input, Silvio! I just opened a new Issue #11 since I feel this would go off-topic here.

SamuelYaron, 2015-02-23:
I updated the simple walker with new links, collision objects and sensors. I also used the share meshes operator and added some joint constraints to the neck. See 2804e0cd835ba0ca0827039e0de81a20086606ca. Is there anything left we need to add to the model for milestone 0.6 ?

SamuelYaron, 2015-02-27:
I also created mass properties, inertias and a motor to the robot

closed 2015-02-27 by SamuelYaron