dfki-ric/phobos #6: Exporting of Boolean values

kavonszadkowski, 2015-02-17:
We have to decide how to treat Boolean custom values. The problem is that Blender interprets Booleans in dictionaries as 1 and 0, not as True and False. We could maybe use strings instead (lowercase *true* and *false* in YAML I think) if we want to make the exported files more readable. Otherwise we have to double-check whether the import in MARS always works correctly (C++ is of course happy with 1s and 0s, but there may be type issues with our ConfigMaps).

SamuelYaron, 2015-03-09:
I have found a very nice way to export our booleans with YAML if we handle them in phobos as strings. You can review it in 1814069135fe95b40341172213f229aebfcca935. There is still the problem to import booleans in phobos again. if we just take the boolean values they will be 0's and 1's again. For that purpose we could (pre-)parse our yaml-files and change all booleans to our string representation, but I think this is rather nasty than good.

SamuelYaron, 2015-03-09:
With 32dc74b8642a1c7b4645ea632d3fbe2d2c8d05ab I found a solution with YAML. Now phobos works with the strings $true and $false. it also can im- and export boolean values properly. Should we think of a way to use this strings in boolean expressions(wrapper class for example) or do we just use them as they are and test for equality with these strings?

closed 2015-03-18 by SamuelYaron