dfki-ric/phobos #7: Type-specific name spaces for import

kavonszadkowski, 2015-02-17:
URDF allows identical names for objects as long as they are of different types, i.e. you could have a visual object called *head* and a collision object called *head*. Phobos does not currently support this, as all object in Blender need to have unique names.

We already have a mechanism allowing objects of different types to be exported using type-specific prefixes that are sliced upon export. We could automatically assign those when importing, either always or only when necessary. Not sure which of the two makes more sense.

SamuelYaron, 2015-02-23:
I think if we want to do it we should do it always in terms of consistency.

kavonszadkowski, 2015-02-27:
Thinking more deeply about editing SMURF scenes in Blender, we have another name space problem whenever robots occur in a scene more than once. Thus this issue should be solved in Milestone 0.7, as there will probably be some interference.

closed 2015-05-12 by SamuelYaron