dfki-ric/phobos #14: Share meshes does not work.

JanPaul73, 2015-02-19:
I had a situation where mesh "doorKnobStart.000" was shared for objects named like "doorKnobEnd.000" and only "doorKnobStart.008.bobj" was exported.
Also this is not easy to undo.
It should be choosable what the shared mesh name will be.

SamuelYaron, 2015-02-20:
The individual naming of shared meshes in now possible(it will be merged into master branch after final testing til end of week). You can undo your mesh sharing instantly with blenders builtin undo mechanism. I didn't find a way to undo the sharing in blender later on yet, but if you just want to export the meshes separately you can delete the objects geometry/sharedMeshFile custom properties and that will do it for exporting out of blender.

kavonszadkowski, 2015-02-20:
Smells like documentation to me. ;)

SamuelYaron, 2015-02-21:
See Issue #9 for the documentation. After testing I merged into master and closed this Issue

closed 2015-02-21 by SamuelYaron

SamuelYaron, 2015-02-23:
This morning I found a way to undo the mesh sharing by accident. I implemented and pushed the operator already and will modify the docs asap.