dfki-ric/phobos #15: Links off center

JanPaul73, 2015-02-19:
Links that are NOT centered in the pivot of the child object (e.g. a door hinge) are still exported to SMURF at the pivot, so that e.g. a door swings around its middle instead of one of its edges.

kavonszadkowski, 2015-02-20:
Child objects (visual/collision) are placed relative to the links, not the other way around, thus the *link skeleton* of a SMURF is only based on the bones in your Blender scene. This sounds to me like it could be a problem with primitive objects' centre points. The primitives we use in Phobos are essentially meshes as everything else in Blender, but are reduced to URDF box/sphere/etc geometries upon export. If the *origin* of the object is not the geometric centre point of the mesh, this will lead to errors. Try Object -> Transform -> Origin to Geometry.
If that doesn't fix it, I need a better description / example scene.

kavonszadkowski, 2015-02-23:
Turns out it is a problem with our use of armatures and bones. Currently, as the position of the link, the origin of the armature is used. Thus the joint axis represented by the bone can be rotated using the *edit bone*, but it can't be shifted. This makes it difficult to alter the position of a link without altering the position of its children.
If we need this functionality, we could make the *edit bone* origin the origin of the link that gets exported. This would lead to problems with the relative positioning, though, as we use [*matrix_local*](https://github.com/rock-simulation/phobos/wiki/Blender#blender-transformation-matrices) for that. It's possible that there is a solution using a different parenting type or so, but that is something we won't implement in any upcoming release.

SamuelYaron, 2015-08-21:
In our last discussion we came up with the idea of using the lots of amatures with one bone way of doing it for simplicity and consistency of the already established workflow. If we ever want to simulate our robots in blender itself we can develop a script that converts our robot into the blender simulation version.

Amudtogal, 2018-08-27:
The current state on this topic is, that we will keep the existing workflow (single bones in armature objects, object origin is the link origin).

Therefore, we can close this for now.

closed 2018-08-27 by Amudtogal