dfki-ric/phobos #17: Where should shared textures between single smurfs of a smurfs-scene go?

JanPaul73, 2015-02-19:
As smurfs for a scene get exported into subdirectories, the probably shared textures have to be copied into each subfolder. A nice feature would be the possibility to choose a relative path for the textures which should be immune to changing between exporting single robots from a scene and the whole scene. This means that the textures should be found by MARS no matter if the single smurfs are loaded from their subdirectories or from a SMURFS-file from their parent directory.

kavonszadkowski, 2015-02-20:
The whole scene export is still quite experimental and needs more fine tuning, especially in terms of configurability. Shared resources between smurfs have so far not been considered, but we'll check what we can include in the upcoming release.

kavonszadkowski, 2015-06-24:
This is something we'll have to talk about in the context of the structured robot export and the scene export using baked robot puppets.

kavonszadkowski, 2015-07-08:
We could make meshes and textures shared between scene objects, however most of the time, this will not be relevant as individual scene elements are stored in their individual repositories anyway. Thus, I will move this to a later milestone, as it's really just cake icing. :)

closed 2018-09-03 by Amudtogal