dfki-ric/phobos #18: Scene export exports sub-smurfs into subfolders, while robot export does not.

JanPaul73, 2015-02-19:
Scene export exports sub-smurfs into subfolders, while robot export does not. It would be better if for example robots were always exported into subdirectories so that no redundant copies are created.

kavonszadkowski, 2015-02-20:
As mentioned in #17, the scene export functionality is still experimental. I'll check how to integrate the two for the coming release.

kavonszadkowski, 2015-02-27:
After restructuring the focus of the coming releases, I moved this feature to Milestone 0.7. In the meanwhile, one can always export individual robots my manually changing the relative export subfolder.

kavonszadkowski, 2015-06-24:
There are two points that in the meanwhile should have rendered this issue obsolete:
- With the new git repository structure for our robot models and the *structured export* feature, the notion of a one-size-fits-all-solution is out of the window as in that case, even the blender file is embedded in a folder for the entire robot.
- Scene compositing including multiple robots should not be the situation in which to edit and thus export individual robots, especially after introduction of the feature that allows "puppets" of existing robots to be used in scene editing.

I'll therefore close this for now. If it continues to be a problem in future workflow, we can reopen it.

closed 2015-06-24 by kavonszadkowski