dfki-ric/phobos #19: Exporting light sources from blender

JanPaul73, 2015-02-20:
Exporting light sources from blender would be a nice addition.

kavonszadkowski, 2015-02-20:
Very true and already planned for a later release, probably 0.7.

kavonszadkowski, 2015-06-11:
What's the status on this issue?

srahms, 2015-06-11:
Right, I should have written a comment on that. Light source export works for lights attached to a robot. Should this be closed, or left open until all light sources can be exported?

kavonszadkowski, 2015-06-17:
Seems like exporting light sources to scenes should be straightforward. Let's discuss this next time we meet and close this issue then.

kavonszadkowski, 2015-07-08:
I made some changes to the implementation. Should work fine and now works for solitary light objects export to SMURF scenes. Probably should be double checked by @srahms.

closed 2015-07-08 by kavonszadkowski