dfki-ric/phobos #24: Sensor creation

kavonszadkowski, 2015-02-27:
Adding sensors is not yet as nice as we would like it. For instance, Joint6DOF sensors need to be added to fixed joints connected to the joints where forces and torques should be measured and sensors attached to that fixed link, cameras are simply anchored to a link while NodeVelocity/NodePosition Sensors can have long lists of nodes they are measuring and don't need to be attached to any link in particular. We need think of a way to make that distinction clearer in the GUI and also document everything properly (though this also relates to MARS documentation itself).

SamuelYaron, 2015-03-05:
There exists some initial documentation in the wiki now (Basic Modelling).

SamuelYaron, 2015-03-05:
I also added support for the Joint6DOF to phobos in a separate branch. it will be merged after testing.

closed 2015-03-09 by SamuelYaron