dfki-ric/phobos #25: "Name Robot" should work on multiple selected robots.

JanPaul73, 2015-03-02:
"Name Robot" should work on multiple selected robots. Then it would be possible to for example export a large number of similar objects that are expected to move independently (e.g. spheres roughly representing water particles).

kavonszadkowski, 2015-03-03:
This is something that relates to us having to complete the scene export. We obviously have to differentiate between the name of the robot *model* and the robot *entity*, the former referring to a robot *class* if you will and the latter to an *instance* of that class.

I think we won't touch this until we've decided how that is going to work, including namespaces for robots. Meanwhile, you can always simply use the batch functionality of the *Edit custom property* operator and simply set "modelname" yourself.

kavonszadkowski, 2015-06-11:
I'm assuming this can be closed, not that #35 is closed?

SamuelYaron, 2015-06-26:
Well the Name Robot Operator does not work on multiple objects. But now we have the namespacing and robot 'puppet' system to create a scene with a number of independent similar robots. So I think this is not needed anymore.

closed 2015-06-26 by kavonszadkowski