dfki-ric/phobos #28: consistent object naming

kavonszadkowski, 2015-03-03:
Phobos allows to define "*type*/name" custom properties, thus enabling the export of custom names for objects (instead of using the default names derived from the Blender object names).
It seems this functionality collides with the lists of objects created and exported for node/joint sensors as well as motors and in some cases one ends up with different names referring to objects in different contexts.
The solution is to always use the custom-defined name property when it exists.

SamuelYaron, 2015-03-11:
I implemented a solution on the branch feature/consistentObjectNaming for the internal object naming consistency. It needs intensive testing!

SamuelYaron, 2015-03-18:
I merged the branch into master today, but it still needs testing and further integration. I will move this Issue to milestone 0.7 to keep it until completion

SamuelYaron, 2015-05-04:
We extended the object naming with the concept of namespacings in phobos. It is essential to use the getObjectName and getObjectByName functions in the utility module. **Never ever use the blender objects name property!**

SamuelYaron, 2015-05-11:
I will assume this Issue closed for now. If someone encounters any problems with names in phobos, just reopen it or create a new one.

closed 2015-05-11 by SamuelYaron