dfki-ric/phobos #31: Update URDF data

kavonszadkowski, 2015-03-03:
It often be the case that a URDF is derived from the CAD files of a robot and imported by Phobos to serve as a starting point to create the SMURF. It will also often be the case that this happens early during design of a robot and as the ongoing construction changes, the URDF exported from CAD changes.

At the moment, this means that someone working with Phobos has to import the updated URDF and manually add all the information such as motors and sensors that was previously added. To avoid such hassle, we should implement an "update URDF" operator that updates the Blender model by moving and reorienting links, updating meshes (optional) etc.

One important issue concerning this functionality is that the names of links imported from URDF will most likely be changed when editing the SMURF. Thus we need a name-mapping functionality to still be able to later update existing, renamed links and joints. This could be done for instance by storing the imported URDF names for links and joints in the respective bones as custom properties that don't get exported or by saving them in a YAML dictionary in a text file (the former is probably easier, but also clutters the model). We might need a rename operator for that, as I we probably cannot catch renaming events in the Blender GUI.

This issue is something that will take time and should be thoroughly debugged, thus I'd propose that we push this to the end of Milestone 0.6.5. If we don't get it done in time, it will be included in 0.7 instead.

SamuelYaron, 2015-10-27:
Partially done in pull request #97. Link creation and mesh data update will be implemented as soon as possible.

Amudtogal, 2018-08-10:
See comment in #97.

closed 2018-08-10 by Amudtogal