dfki-ric/phobos #33: URDF export: fixed order of elements

kavonszadkowski, 2015-03-03:
Phobos does not adhere to a consistent (e.g. alphabetical) order of URDF elements on export which makes it hard to compare different versions with one another using diff tools such as Meld or Diffuse and which makes committing to git repositories etc. that much harder. We should fix this.

kavonszadkowski, 2015-06-17:
What's the status on this issue? The feature is starting to get really important, since we have begun moving some of our models in repositories.

srahms, 2015-07-03:
The general sorting has been implemented for some time now but there were still some problems that have now been fixed. The behaviour is now as follows:
- generally, during export, links, joints, collisions, visuals and materials are being sorted alphabetically
   - if URDF files were imported previous to export, elements that existed in these files retain the order they had there, while new elements are appended in alphabetical order
- elements of the SMURF YAML files are sorted alphabetically

kavonszadkowski, 2015-07-20:
Doesn't seem to be fixed for lists such as the parameters of joint limits - what kind of solution for that could we come up with?

srahms, 2015-07-22:
I actually made a mistake with the handling of limits and completely overlooked all the occurrences of the other tags with multiple attributes. I could sort all of these at export, but the real problem is keeping the order of an imported URDF.
Since attribute order is defined in XML as not significant, the employed parser returns parsed attributes as dictionary.
So, as I see it, there are three ways to solve this:
1. checking the attribute order manually in an imported file (dirty, but consequent)
2. if a file was imported, simply trust in Python to order dictionary keys consistently, which it actually does in most cases, even though key order is not defined (unsafe, but it currently works pretty well like that)
3. always sort multiple attributes at export (consequent, but ugly diffs to old/non-phobos-generated URDFs)
I would tend to solution 1, as it is the only one whose downside only affects code efficiency and elegance.
Other suggestions?

srahms, 2015-07-22:
Solution 4: Write a parser class that inherits from the XMLTreeBuilder and use OrderedDict for the attributes. An example of doing this does actually exist here:
http://stackoverflow.com/questions/2741480/can-elementtree-be-told-to-preserve-the-order-of-attributes

srahms, 2015-07-22:
Solution 3 has been implemented with attribute order based on the URDF specification.

closed 2015-07-22 by srahms