dfki-ric/phobos #36: Multiple parents for one object

FrankIhle, 2015-03-05:
For beeing able to construct closed loops of joints and bodys (parallel kinematics), objects have to be adjustable to have more than one parent, otherwise a wrong physical representation for this purpose is generated in Phobos

kavonszadkowski, 2015-03-05:
While this is true in principle, URDF (the format underlying SMURF and therefore robots created using Phobos) does not support parallel kinematics. There is the concept of *mimic joints* which to my knowledge hasn't been fully implemented in all related software tools, yet.

Historically, we have solved this issue by creating plugins for MARS calculating the forces for the parallel kinematics and moving the related un-parented visual-objects accordingly.

We might revisit this in Milestone 0.9.

SamuelYaron, 2015-11-03:
How is this Issues status, @kavonszadkowski ?

closed 2017-03-02 by kavonszadkowski