dfki-ric/phobos #37: Joint limits translate to strange values in blender bone constraints.

JanPaul73, 2015-03-05:
Joint limits translate to strange values in blender bone constraints. A joint limit like for example max 90° is stored in the blender bone constraints as 5156.62d, which makes no sense to me, even when goint from rad to degrees or back.

However, it is correctly exported as max 90° into URDF and motors.yml, however, I can still move a door with min 0° and max 90° by more than 90° (which is probably more a problem on the mars side).

kavonszadkowski, 2015-03-05:
90 rad = 5156.62°
If you want to type in the joint limits as degrees, you have to check the *degrees* box of the operator.
What you therefore export is a max value of 90 radians, which no doubt results in joints that can turn a lot.

closed 2015-03-05 by kavonszadkowski

JanPaul73, 2015-03-05:
Ah right, just recognized that. Additionally, I had the min/max values mixed up. Now it works, thanks :-).