dfki-ric/phobos #38: Export of meshes using modifiers.

JanPaul73, 2015-03-05:
Exporting meshes with modifiers tends to crash Blender and MARS when using .obj and only MARS when using .bobj. Applying the modifiers before solves the problem, so when exporting (which to my knowledge includes making a copy of the object), applying all the modifiers to the copy should solve the problem.

Another problem is probably active here: The copy is moved to [0,0,0], right? If it uses a boolean operation with another object "A" or is for example deformed by a lattice "A", then the object will get corrupted by moving it away from its current position (as the other object "A" will stay in place). So the object should be
1. copied in place
2. get all modifierts applied
3. be set to [0,0,0] or wherever necessary
4. then get exported

kavonszadkowski, 2015-03-05:
Not a pressing issue right now. Feel free to create a pull request, though.

JanPaul73, 2015-03-31:
This is fixed for BOBJ now (see my pull request), but still open for OBJ.

SamuelYaron, 2015-10-27:
Done for obj and stl in pull request #98 .

closed 2015-10-27 by SamuelYaron