dfki-ric/phobos #40: Masses are always exported as 0.

JanPaul73, 2015-03-10:
Masses are seemingly ignored, when setting them (in visual, collision and inertial object) to e.g. 2, the mass of the node stays 0, the density stays always 1.

kavonszadkowski, 2015-03-11:
I'm afraid I cannot reproduce the problem you describe. Mass export has been working for quite a while now and when I just checked, is working still. Please describe the specific scenario in which mass export does not work, preferably a minimum example.

I can imagine one source of confusion: Depending on how many visual and collision objects are parented to a link, the link might become split up when imported in MARS into several nodes, in which case mass properties are may not be the same as expected in every "child node".

JanPaul73, 2015-03-11:
Hi,

I have attached a minimal example. I exported the scene as Robot Model and SMURF
Scene. In both cases, I cannot find any mass properties in the result files.

> Kai von Szadkowski <notifications@github.com> hat am 11. März 2015 um 09:22
> geschrieben:
> 
> 
>  I'm afraid I cannot reproduce the problem you describe. Mass export has been
> working for quite a while now and when I just checked, is working still.
> Please describe the specific scenario in which mass export does not work,
> preferably a minimum example.
> 
>  I can imagine one source of confusion: Depending on how many visual and
> collision objects are parented to a link, the link might become split up when
> imported in MARS into several nodes, in which case mass properties are may not
> be the same as expected in every "child node".
> 
> <https://github.com/rock-simulation/phobos/issues/40#issuecomment-78221754> .

--
Dr. Jan Paul
Simulation

Hauptgeschäftsstelle Standort Bremen:
DFKI GmbH
Robotics Innovation Center
Robert-Hooke-Straße 1
28359 Bremen, Germany

Tel.: +49 421 178 45-6618
Zentrale: +49 421 178 45-0
Fax: +49 421 178 45-4150 (Faxe bitte namentlich kennzeichnen)
E-Mail: jan.paul@dfki.de

Weitere Informationen: http://www.dfki.de/robotik
-----------------------------------------------------------------------
Deutsches Forschungszentrum fuer Kuenstliche Intelligenz GmbH
Firmensitz: Trippstadter Straße 122, D-67663 Kaiserslautern
Geschaeftsfuehrung: Prof. Dr. Dr. h.c. mult. Wolfgang Wahlster
(Vorsitzender) Dr. Walter Olthoff
Vorsitzender des Aufsichtsrats: Prof. Dr. h.c. Hans A. Aukes
Amtsgericht Kaiserslautern, HRB 2313
Sitz der Gesellschaft: Kaiserslautern (HRB 2313)
USt-Id.Nr.: DE 148646973
Steuernummer: 19/673/0060/3
-----------------------------------------------------------------------

kavonszadkowski, 2015-03-13:
The link-inertial in your example contains *mass* and not *inertial/mass*, hence the malfunction.
Only link-inertials get assigned the correct property automatically using the relevant operator.

closed 2015-03-13 by kavonszadkowski