dfki-ric/phobos #42: Texture export uses wrong name field

JanPaul73, 2015-03-16:
Texture export uses the name set in Object->Texture->Image->First Line ("Image Datablock...") instead of the filename set in "Source" below. When the source image name changes (by selecting a new image with a different name), the "Image Datablock" name above stays the same and references a wrong or deleted image.

kavonszadkowski, 2015-06-17:
As we are currently defining a structure to put robot models in repositories, we will introduce, similar to the "meshes" folder for meshes, a "textures" folder for textures. This should be fixed along with the naming issues.

SamuelYaron, 2015-11-17:
In its current form phobos uses the correct filename and the textures blender intern name (e.g. Material.003) as the name. I think this is the expected behavior. I will close this Issue.

closed 2015-11-17 by SamuelYaron