dfki-ric/phobos #43: Pros and cons with obj and bobj export options

JanPaul73, 2015-03-16:
When using the bobj-export option, all modifiers are applied before exporting. However, then the UV-coordinates get messed up. Even if the modifier is manually applied before. The texture is repeated over e.g. new faces added by a subdivision modifier.

When exporting with the obj-option however, the UVs stay correct, even if e.g. new faces where generated with a subdivision modifier. However, the modifiers are not automatically applied, so that the results of the modifiers are only present in the obj if they have been manually applied before.

The best of both worlds would be that both formats keep the UVs intact and apply all modifiers before export.

kavonszadkowski, 2015-03-17:
I agree with your summary.

Interesting that the modifiers are applied with the bobj-option. I haven't implemented that myself but I remember a lot of black magic going on there, manually copying mesh data. Probably somewhere in that black magic the UV-coordinates don't get copied properly.

I'll put this in Milestone 1.0, as the points you mention are more of a cosmetic/minor usability issue.

kavonszadkowski, 2015-06-11:
Revisiting this, have we fixed that with merging your pull request a while ago, Jan?

JanPaul73, 2015-06-11:
For BOBJ yes, but not for OBJ.

Amudtogal, 2018-09-03:
If this becomes relevant again, please reopen this issue.

As we do not have a bobj export in Phobos, this is not a problem right now...

closed 2018-09-03 by Amudtogal