dfki-ric/phobos #53: Relative positions of individual robots in scene wrong.

JanPaul73, 2015-04-20:
In a scene I have multiple robots which are positioned differently to one another in the MARS simulation compared to the original scene in Blender. Only the position of root links is wrong, the position of child links within a robot is OK.

I am currently looking into the issue. Might be something relating to the root links being a special case as their positions are not stored in joints but in the scene's smurfs-file. (Although everything seems to be correct so far.)

kavonszadkowski, 2015-04-21:
Info for everyone who is listening: We just spoke on the phone and Jan told me that it seems that Phobos writes out the data correctly and the error lies with MARS.

kavonszadkowski, 2015-06-11:
Apparently this comes down to an issue in MARS where the origin of a root link might be moved to the COM of the inertial. Either way, it is does not seem to be a problem with Phobos.

closed 2015-06-11 by kavonszadkowski