dfki-ric/phobos #54: Unable to import .scn file directly

bergatt, 2015-04-21:
Hi @kavonszadkowski,

when trying to load an old .scn file phobos crashes saying:

```
Unknown model format, aborting import...
Traceback (most recent call last):
  File "/Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos/importer.py", line 2070, in execute
    importer.parseModel()
UnboundLocalError: local variable 'importer' referenced before assignment
```

I had to extract the .scn into a folder and was then able to load the .scene
but it shouldn't be too hard to also support the unpacking the .scn before
importing the .scene files.

Best regards,
Constantin

srahms, 2015-04-22:
Importing MARS scenes from .scn files is now supported.

bergatt, 2015-04-24:
Hi,

I just got an error while testing the new feature. When starting blender on Mac OS X with the help of spotlight and trying to import a zipped MARS scene, I got the following error:
```
Traceback (most recent call last):
  File "/Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos/importer.py", line 2130, in execute
    importer.createBlenderModel()
  File "/Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos/importer.py", line 788, in createBlenderModel
    self.createLink(link)
  File "/Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos/importer.py", line 615, in createLink
    self.createGeometry(visual, 'visual')
  File "/Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos/importer.py", line 492, in createGeometry
    os.mkdir(tmp_dir_name)
PermissionError: [Errno 13] Permission denied: 'phobos_magic_zip_tmp_dir'

location: <unknown location>:-1
```
It's working if I start blender from the command line.

Regards,
Constantin

srahms, 2015-04-24:
Hi,

this  error was due to me having my Blender executable in my home directory and thus not realising the need for write permissions. This should now be fixed.

Regards,
Stefan

bergatt, 2015-04-27:
Hi Stefan,

thanks for the quick fix.

It now works quite nicely.

Thanks a lot!

Regards,
Constantin

closed 2015-04-27 by bergatt