dfki-ric/phobos #55: Install script won't find phobos installation on Mac OS X and first installation won't copy definitions folder

bergatt, 2015-04-24:
Hi,

the last days I tried to import a few MARS scenes into blender with phobos and did find bugs or errors within the installation scripts.

First of all, when installing phobos the first time the definitions folder is not copied, what is done when updating the installation.

Second, the test if phobos is already installed on Mac OS X is always failing because
```
if [ -d $phobospath ]
```
is returning an error.
```
install_phobos.sh: line 91: [: /Users/bergatt/Library/Application: binary operator expected
```

I fixed my problems by applying the following changes to the script:
```
diff --git a/install_phobos.sh b/install_phobos.sh
index 17649d5..9887199 100755
--- a/install_phobos.sh
+++ b/install_phobos.sh
@@ -91,8 +91,8 @@ installPhobos() {
     if [ -d $phobospath ]
     then
         echo "Phobos installation found and updated."
-        cp *.py $phobospath
-        cp -R definitions $phobospath
+        cp *.py "$phobospath"
+        cp -R definitions "$phobospath"
         installed=1
     else
         echo "Phobos folder does not exist, create phobos folder in $addonpath ? (y/n)"
@@ -101,6 +101,7 @@ installPhobos() {
             y|Y )
                 mkdir -p "$phobospath"
                 cp *.py "$phobospath"
+                cp -R definitions "$phobospath"
                 installed=1
                 echo "Copied phobos to $phobospath"
                 ;;
```

Thought it would be of interest to you guys.

Best regards,
Constantin

SamuelYaron, 2015-05-04:
Thank you very much for sharing your solution Constantin,
I cannot test the Mac OS X related problem with the if statement in line 91 due to the lack of a OS X machine, but I applied your changes to the install script.

SamuelYaron, 2015-05-04:
Malte and I cannot reproduce your error with the if statement on his Mac OS X. We're assuming that it is the same problem your other changes resulting from. Can you try to reproduce your error with putting the $phobospath in the if statement in ' " '? If it solves the problem I will include it in the install script as well.

bergatt, 2015-05-15:
Hi,

I just tried it out and it didn't work.

```
bergatt@bergatt-macpro ~/Repositories/mars/phobos  [master]$ ./install_phobos.sh 
Found install config file.
Check for phobos installation...
./install_phobos.sh: line 91: [: /Users/bergatt/Library/Application: binary operator expected
Phobos folder does not exist, create phobos folder in /Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons ? (y/n)
n
No folder for phobos created
Cannot install YAML, because Phobos wasn't installed.
```

When  adding ' " ' to line 91, it works for me.

```
bergatt@bergatt-macpro ~/Repositories/mars/phobos  [master]$ ./install_phobos.sh 
Found install config file.
Check for phobos installation...
Phobos installation found and updated.
YAML package was copied into /Users/bergatt/Library/Application Support/Blender/2.73/scripts/addons/phobos.
```

Kind regards,
Constantin

closed 2015-05-20 by SamuelYaron