dfki-ric/phobos #59: The problem with the EDIT mode in blender

SamuelYaron, 2015-05-11:
I encountered the EDIT mode problem we already had again. During the joint creation while importing an URDF I got this error:

```
Creating joints...
Traceback (most recent call last):
  File "/home/oschwiegert/.config/blender/2.71/scripts/addons/phobos/importer.py", line 2129, in execute
    importer.createBlenderModel()
  File "/home/oschwiegert/.config/blender/2.71/scripts/addons/phobos/importer.py", line 801, in createBlenderModel
    self.createJoint(joint)
  File "/home/oschwiegert/.config/blender/2.71/scripts/addons/phobos/importer.py", line 691, in createJoint
    bpy.ops.object.mode_set(mode='EDIT')
  File "/home/oschwiegert/tools/Blender-2.71/2.71/scripts/modules/bpy/ops.py", line 188, in __call__
    ret = op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties:  enum "EDIT" not found in ('OBJECT')
```

In my opinion this is a crucial bug we have to get rid of asap.

SamuelYaron, 2015-05-11:
I discovered that this "bug"only occurs if I try to import an URDF file where two ore more objects share the same name. Does someone has any ideas?

kavonszadkowski, 2015-05-12:
Sounds right. Since we are now using specific custom properties for all object names, we can name objects with type-specific prefixes as proposed in #7.

SamuelYaron, 2015-05-12:
Should be fixed with the closing of Issue #7

closed 2015-05-12 by SamuelYaron