dfki-ric/phobos #60: Scaled objects exported to SMURF are scaled in MARS too

ChrisStil, 2015-05-21:
When someone wants to export a robot model to SMURF with Phobos, and some meshes or links have a scale value other than 1.0 (in the Transform panel), The model loaded in MARS is scaled too. That can be confusing because the model that we see in Blender can be completely different from the model seen in MARS.

[image]

For the moment, selecting all objects and using the shortcut CTRL-A -> Scale before the export is the only solution to this problem.

kavonszadkowski, 2015-05-27:
From the top of my hat, I'm not sure whether this is solely an issue with Phobos, but (or only) a problem in MARS. We'll have to check that.

Amudtogal, 2018-08-15:
I assume this has been fixed by now.
If other issues arise, please reopen this Issue.

closed 2018-08-15 by Amudtogal