dfki-ric/phobos #62: Could't import phobos.collisions

PierreJac, 2015-06-23:
Hello,
I tried to install phobos on blender 2.69 (with Ubuntu 14.04), however  I always get two errors.
First of all I can't import the module defs (even with the help of the issue #55 but as I'm not really at ease with linux it may be my fault)
Second, I sometimes get the errors : "import error : No module named phobos.collisions" (line 121 of __init__.py). I checked into the src folder and it seems that collisions.py is missing.

Can you help me with these issues ? Your addon seems really awsome and I really want to try it !

Best regards,
Pierre

kavonszadkowski, 2015-06-23:
Hi Pierre,
sorry for the inconvenience! It took only 1 minute for another person to report that error, so I can already verify the error is on our side, not yours.

SamuelYaron, 2015-06-23:
Wit the commit ea99909 the problem with the collision import should be fixed. The refactoring might be the reason for the problems. On my machine the other import error occurs only when I decide to use the distributed yaml package with phobos. This needs further investigation.

SamuelYaron, 2015-06-26:
We have an Issue ( Issue #63 ) describing the last unsolved problem in this Issue very well. I will close this one now.

closed 2015-06-26 by SamuelYaron