dfki-ric/phobos #63: Error while  installing phobos

ChrisStil, 2015-06-26:
Hi, I habe the following problem while updating phobos : 

Version of Blender : tested on 2.69 and 2.74
Os : Ubuntu 14.04 (tested on 2 different computers)
commit version of phobos : 65cf79e7917b93144a5a72cc6bb67b44211eaf1a (but first seen on ea999096f812f278742f237c18d85e4fe668d117)

I runned git pull/clone in the phobos directory and runned the install_phobos.sh script. Everything worked fine in the installation. I used the yaml version distributed with phobos.

Wenn I want to activate the plugin in blender in the user preferences,

[image]

I get the following output in the terminal (and no layout of phobos is shown): 

./blender
connect failed: No such file or directory
found bundled python: /home/swall/Software/blender-2.74/2.74/python
Could not find yamlpath.conf
Using distributed package instead!
Traceback (most recent call last):
  File "/home/swall/Software/blender-2.74/2.74/scripts/modules/addon_utils.py", line 306, in enable
    mod = __import__(module_name)
  File "/home/swall/.config/blender/2.74/scripts/addons/phobos/__init__.py", line 62, in <module>
    import yaml
  File "/home/swall/.config/blender/2.74/scripts/addons/phobos/yaml/__init__.py", line 8, in <module>
    from .loader import *
  File "/home/swall/.config/blender/2.74/scripts/addons/phobos/yaml/loader.py", line 8, in <module>
    from .constructor import *
  File "/home/swall/.config/blender/2.74/scripts/addons/phobos/yaml/constructor.py", line 8, in <module>
    import collections, datetime, base64, binascii, re, sys, types
  File "/home/swall/Software/blender-2.74/2.74/python/lib/python3.4/collections/__init__.py", line 11, in <module>
    from operator import itemgetter as _itemgetter, eq as _eq
ImportError: cannot import name 'itemgetter'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/swall/Software/blender-2.74/2.74/scripts/startup/bl_operators/wm.py", line 1741, in execute
    mod = addon_utils.enable(self.module, default_set=True, handle_error=err_cb)
  File "/home/swall/Software/blender-2.74/2.74/scripts/modules/addon_utils.py", line 314, in enable
    handle_error()
  File "/home/swall/Software/blender-2.74/2.74/scripts/startup/bl_operators/wm.py", line 1736, in err_cb
    import traceback
  File "/home/swall/Software/blender-2.74/2.74/python/lib/python3.4/traceback.py", line 3, in <module>
    import linecache
  File "/home/swall/Software/blender-2.74/2.74/python/lib/python3.4/linecache.py", line 10, in <module>
    import tokenize
  File "/home/swall/Software/blender-2.74/2.74/python/lib/python3.4/tokenize.py", line 29, in <module>
    import collections
  File "/home/swall/Software/blender-2.74/2.74/python/lib/python3.4/collections/__init__.py", line 11, in <module>
    from operator import itemgetter as _itemgetter, eq as _eq
ImportError: cannot import name 'itemgetter'

SamuelYaron, 2015-06-26:
Hi ChrisStll, we encountered this error in Issue #62
There is a problem with the use of the distributed yaml package. A workaround is to install yaml for python3 (often the package name is python3-yaml) and install phobos with the python library option.

SamuelYaron, 2015-08-21:
Hi ChrisStil are you still encountering errors while installing phobos? It would be very helpful if you try it again and report your results.

SamuelYaron, 2015-09-29:
I will close this Issue due to inactivity. If one encounters this problem again feel free to reopen this Issue or create a new one.

closed 2015-09-29 by SamuelYaron