dfki-ric/phobos #64: ImportError: cannot import name 'defs'

130s, 2015-06-27:
Not sure how this is related to #56 but while I'm trying to activate phobos, I get the following error.

```
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 298, in enable
    mod = __import__(module_name)
  File "/home/rosfettucini/.config/blender/2.69/scripts/addons/phobos/__init__.py", line 67, in <module>
    from . import defs
ImportError: cannot import name 'defs'
```

```
$ dpkg -p python3 python3-yaml|grep Ver
Version: 3.4.0-0ubuntu2
Version: 3.10-4ubuntu0.1
$ more /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.2 LTS"
$ ./install_phobos.sh 
Enter your blender version (e.g. '2.71')
2.69
Do you want to use your python3 yaml library or do you want to use the yaml version
    distributed with phobos? (p - use python3 library / d - use distributed package)
p
Please enter the command to run your python3 binary (e.g python3 or /usr/bin/python3)
python3
Check for phobos installation...
Phobos folder does not exist, create phobos folder in /home/n130s/.config/blender/2.69/scripts/addons ? (y/n)
y
Copied phobos to /home/n130s/.config/blender/2.69/scripts/addons/phobos
Checking for yamlpath.conf
yamlpath.conf created and copied
```

130s, 2015-06-27:
(Rewriting my issue report.)

Not sure how this is related to #56 but while I'm activating phobos, I get the following error when I tick the checkbox on User Preference window, on the console? textfield at upper-right on Blender:

```
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 298, in enable
    mod = __import__(module_name)
  File "/home/rosfettucini/.config/blender/2.69/scripts/addons/phobos/__init__.py", line 67, in <module>
    from . import defs
ImportError: cannot import name 'defs'
```

Also, I often get the following error, but not always. Maybe only the first attempt of ticking the checkbox.
```
NameError: name 'yaml' is not defined
```

[image]

```
$ dpkg -p python3 python3-yaml|grep Ver
Version: 3.4.0-0ubuntu2
Version: 3.10-4ubuntu0.1
$ more /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.2 LTS"
$ ./install_phobos.sh 
Enter your blender version (e.g. '2.71')
2.69
Do you want to use your python3 yaml library or do you want to use the yaml version
    distributed with phobos? (p - use python3 library / d - use distributed package)
p
Please enter the command to run your python3 binary (e.g python3 or /usr/bin/python3)
python3
Check for phobos installation...
Phobos folder does not exist, create phobos folder in /home/n130s/.config/blender/2.69/scripts/addons ? (y/n)
y
Copied phobos to /home/n130s/.config/blender/2.69/scripts/addons/phobos
Checking for yamlpath.conf
yamlpath.conf created and copied
```

ChrisStil, 2015-06-28:
I had the same Problem : A " sudo apt-get install python3-yaml " and a new git clone of phobos solved this Problem on my Computer.

SamuelYaron, 2015-06-29:
Hi 130s, I tried to reproduce your error with the same parameters but I couldn't. On my machine the import works fine. Which commit of this repository do you use?

130s, 2015-06-30:
@SamuelYaron thx for taking a look. I used 65cf79e7917b93144a5a72cc6bb67b44211eaf1a
Author: Ole Schwiegert <ole.schwiegert@uni-bremen.de>
Date:   Fri Jun 26 12:38:53 2015 +0200

130s, 2015-06-30:
And I did `sudo apt-get install --reinstall python3-yaml`, which didn't help.

kavonszadkowski, 2015-07-20:
We couldn't really reproduce the error. Are there any updates on this issue on your side? Have you tried with the latest version of Phobos?
It didn't seem to make a difference in our tests, but have you also tried using the latest version of Blender (2.75)? You can download it from Blender's website and unzip it to a local folder, and you'll have to change the install path for Phobos.

SamuelYaron, 2015-08-21:
Hi 130s maybe I solved your Issue with the import of defs by accident. May you try the current master branch and report your results? (You still have to use the python3-yaml package and not the distributed one due to an unresolved error with the yaml import)

130s, 2015-08-21:
Ok, this particular problem seems to be fixed for me. Thanks!
(I still face another issues though.)

closed 2015-08-21 by 130s