dfki-ric/phobos #66: URDF joints: proper export

kavonszadkowski, 2015-07-20:
We should implement the default use of x-axis as the joint axis unless somehow explicitly specified otherwise. This could be done by checking the orientation offset between armature and bone - which is of course only possible if we don't switch to a one-armature-many-bones structure; in that case we could add another bone attached to the actual link frame that indicates the axis, this was the original implementation.

This relates to #15 and #31, so we should be careful to take these into account.

SamuelYaron, 2015-08-21:
As discussed in Issue #15 we decided to keep the many armatures one bone way to do it. So this Issue can be solved now.

SamuelYaron, 2015-09-29:
I have a solution where we just rotate a  bone 90° around the y-axis and apply this rotation. This can be done at bone creation (Implement this behaviour in the createBone function) and on existing models as well. In the second case we have to remember the current rotation, reset it and then apply this 90° rotation. After it we can add the saved rotation to the object. This can be achieved via the following script:

```python
import math, bpy

links = filter(lambda o: o.phobostype == "link", bpy.context.selected_objects)

bpy.ops.object.select_all(action='DESELECT')

for link in links:
    rot = tuple([v for v in link.rotation_euler])
    link.rotation_euler = (0,math.pi/2,0)
    link.select = True
    bpy.context.scene.objects.active = link
    bpy.ops.object.transform_apply(location=False, rotation=True, scale=False)
    link.select = False
    link.rotation_euler = rot
```

There is one very important limitation to this script. Without any rotation, the bones tail has to point towards the positive z axis. In the mantis for example I found a lot of bones pointing towards the negative z axis.

On bone creation though this method works well.

An exported joint then looks like this.

```xml
<joint name="extremity5/4" type="revolute">
      <origin xyz="-0.0045379 -0.0002668 -3.99e-05" rpy="0 1.5707963 0"/>
      <parent link="extremity5/5"/>
      <child link="extremity5/4"/>
      <axis xyz="1.0 0 0"/>
      <limit lower="-3.1400001" upper="3.1400001" effort="28.0" velocity="3.1400001"/>
    </joint>
```

This method allows us to set individual joint axis as the model creator likes.

SamuelYaron, 2015-11-17:
The new standard behavior is the x-axis as the joint axis. It is also possible to use the y and z axis as well. The only thing missing is an automatic way to recalculate already existing models. They're still working though and using the z axis.

AlCap23, 2019-01-28:
Cannot confirm or disconfirm validity of request any more. Closed.

closed 2019-01-28 by AlCap23