dfki-ric/phobos #67: installation failes (yamlpath.conf not found)

goldhoorn, 2015-07-28:
I run the installation script in debian stable with blender and python-yaml installed i got the following error:

```
Enter your blender version (e.g. '2.71')
2.72
Do you want to use your python3 yaml library or do you want to use the yaml version
    distributed with phobos? (p - use python3 library / d - use distributed package)
p
Please enter the command to run your python3 binary (e.g python3 or /usr/bin/python3)
/usr/bin/python3
Check for phobos installation...
Phobos folder does not exist, create phobos folder in /home/goldhoorn/.config/blender/2.72/scripts/addons ? (y/n)
y
cp: das angegebene Ziel „/home/goldhoorn/.config/blender/2.72/scripts/addons/phobos“ ist kein Verzeichnis
cp: das Verzeichnis „/home/goldhoorn/.config/blender/2.72/scripts/addons/phobos“ kann nicht angelegt werden: Datei oder Verzeichnis nicht gefunden
Copied phobos to /home/goldhoorn/.config/blender/2.72/scripts/addons/phobos
Checking for yamlpath.conf
YAML module found!
cp: reguläre Datei „/home/goldhoorn/.config/blender/2.72/scripts/addons/phobos/yamlpath.conf“ kann nicht angelegt werden: Datei oder Verzeichnis nicht gefunden
yamlpath.conf created and copied
```

goldhoorn, 2015-07-28:
solved by creatign the directoy manually, please fix the install script
```
$ mkdir -p /home/goldhoorn/.config/blender/2.72/scripts/addons/phobos
$ ./install_phobos.sh 
Found install config file.
Check for phobos installation...
Phobos installation found and updated.
Checking for yamlpath.conf
YAML module found!
yamlpath.conf created and copied
```

closed 2015-07-28 by SamuelYaron

kavonszadkowski, 2015-07-28:
Thanks, Matthias.