dfki-ric/phobos #70: URDF importer does not work correctly

goldhoorn, 2015-07-28:
I tried to import models from the following file:

https://github.com/goldhoorn/p2os/blob/master/p2os_urdf/defs/model.urdf

however the collision's seems valid but all visuals are misplaced.

(same is true for other urfd files from this package)

kavonszadkowski, 2015-07-30:
I reckon we'll first have to check whether the URDFs you mentioned are valid to begin with.

goldhoorn, 2015-07-30:
It's auto generated by some script (what does not mean that is is 
correct) but is 'common'
however you can find under the link i already posted.

-- 
  Dipl.-Inf. Matthias Goldhoorn
  Space and Underwater Robotic

  Universität Bremen
  FB 3 - Mathematik und Informatik
  AG Robotik
  Robert-Hooke-Straße 1
  28359 Bremen, Germany
  
  Zentrale: +49 421 178 45-6611
  
  Besuchsadresse der Nebengeschäftstelle:
  Robert-Hooke-Straße 5
  28359 Bremen, Germany
  
  Tel.:    +49 421 178 45-4193
  Empfang: +49 421 178 45-6600
  Fax:     +49 421 178 45-4150
  E-Mail:  matthias.goldhoorn@informatik.uni-bremen.de

  Weitere Informationen: http://www.informatik.uni-bremen.de/robotik

closed 2017-03-02 by kavonszadkowski