dfki-ric/phobos #79: ImportError: No module named 'phobos.robotupdate'

130s, 2015-08-21:
After #64 is resolved for me, I still see the following error when I try to enable Phobos by ticking the checkbox in User Preference.

```
$ dpkg -p python3 python3-yaml|grep Ver
Version: 3.4.0-0ubuntu2
Version: 3.10-4ubuntu0.1
$ ./install_phobos.sh 
Found install config file.
Check for phobos installation...
Phobos folder does not exist, create phobos folder in /home/rospasta/.config/blender/2.69/scripts/addons ? (y/n)
y
Copied phobos to /home/rospasta/.config/blender/2.69/scripts/addons/phobos
Checking for yamlpath.conf
YAML module found!
yamlpath.conf created and copied

$ blender --version
Color management: using fallback mode for management
Blender 2.69 (sub 0)
	build date: 2014-03-22
	build time: 21:47:21
:

$ blender &
[1] 9342
~/ROS/indigo_trusty/cws_viz/src/rock-simulation/phobos$ Color management: using fallback mode for management
ndof: unknown Logitech product c404
Importing yaml module
Using following folder for defs: /home/rospasta/.config/blender/2.69/scripts/addons/phobos/definitions
Parsing YAML files for updating defs
Parsing defaultSensors.yml
Parsing defaultMotors.yml
Parsing defaultJoints.yml
Parsing defaultDictConstraints.yml
Importing utils module
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 298, in enable
    mod = __import__(module_name)
  File "/home/rospasta/.config/blender/2.69/scripts/addons/phobos/__init__.py", line 119, in <module>
    import phobos.links, phobos.robotdictionary, phobos.controllers, \
  File "/home/rospasta/.config/blender/2.69/scripts/addons/phobos/operators/editing.py", line 39, in <module>
    import phobos.robotupdate as robotupdate
ImportError: No module named 'phobos.robotupdate'

Importing yaml module
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 298, in enable
    mod = __import__(module_name)
  File "/home/rospasta/.config/blender/2.69/scripts/addons/phobos/__init__.py", line 67, in <module>
    import phobos.defs as defs
AttributeError: 'module' object has no attribute 'defs'
```

SamuelYaron, 2015-08-21:
Thank you very much for your input. We changed our structure in the last few month and remaining artifacts on our machines preventing errors occuring on fresh installations. The import of robotupdate is one of this prevented errors. So the first exception should be solved now. I can not reproduce your second exception despite the fresh and clean installation I made. Maybe the error was caused by the first one. Please try it again. If the error still exists we may have to deeply investigate into blenders python module loading structure to prevent all the import errors occuring from time to time.

130s, 2015-08-21:
https://github.com/rock-simulation/phobos/commit/8aeed581caaf35f50708d4e182c1a9c4ccdcd3b4 seems to be working with re-installed blender (which I'm not sure if it was necessary). Now `phobos` is working for me. Thanks.

closed 2015-08-21 by 130s

SamuelYaron, 2015-08-21:
This sounds very good. We hope phobos will be useful for you. If you encounter any problems let us know.