dfki-ric/phobos #105: install error on blender 2.76b in Ubuntu14.04/Windows7-64b

pgaudillere, 2016-01-10:
Hi,
I tried to install phobos on Blender 2.76b, and I got the folowing error:
[image]
any idea? Do I have to try on Blender 2.71?

best regards,

Pierre

pgaudillere, 2016-01-10:
[image]

kavonszadkowski, 2016-01-11:
The problem seems to arise when Phobos tries to import its predefined components, such as sensors. I myself am running on Ubuntu 14.04 and Blender 2.76b, so that shouldn't be the issue, but I see from the error message that error occurred on Windows?

pgaudillere, 2016-01-11:
Yes I tried to import the 0.7 zip release from github on both windows and linux, with no success. But the install script on linux from the git snapshot works. I wonder if there are missing files in the zip file..
Now I have difficulties to export my robot to urdf (a bit too complicated for a first try I think), but it is another subject.. Thanks for your help, and this great addon, I was thinking about doing one myself, but yours looks great!

closed 2016-05-13 by kavonszadkowski

oschumac, 2017-05-13:
Hi i had the same prob. I have copied the setting folder to the script folder now everything runs fine. CU