dfki-ric/phobos #106: URDF <transmission> tag missing

alextoind, 2016-02-12:
If I attach a motor to a link and I export to URDF, I cannot find the expected `<transmission>` tag, even if its data appears correctly in the YAML dictionary:

    motors:
      pelvis_roll: {d: 10.0, i: 0.01, joint: pelvis_roll, maxEffort: 100.0, maxSpeed: 10.0, maxValue: 0.25, minValue: -0.25, name: pelvis, p: 100.0, type: PID}

Is this ok?

(I use Blender 2.76b and Phobos 0.7)

kavonszadkowski, 2016-02-12:
According to [the URDF page on transmission elements](http://wiki.ros.org/urdf/XML/Transmission), "only the ros_control project uses this transmission elements". Thus we haven't implemented support for them yet, since there seems to be an ongoing discussion.

alextoind, 2016-02-15:
Oh, I see. Thank you for the reply.

closed 2016-05-13 by kavonszadkowski