dfki-ric/phobos #107: YAML install problem on Ubuntu 16.04/Blender 2.77

yamokosk, 2016-05-10:
Here is the error I get when trying to activate phobos. I used YAML distributed with phobos.

> $ ./blender
> connect failed: No such file or directory
> found bundled python: /home/yamokosk/bin/blender-2.77a-linux-glibc211-x86_64/2.77/python
> Could not find yamlpath.conf
> Using distributed package instead!
> Traceback (most recent call last):
>   File "/home/yamokosk/bin/blender-2.77a-linux-glibc211-x86_64/2.77/scripts/modules/addon_utils.py", line 330, in enable
>     mod = __import__(module_name)
>   File "/home/yamokosk/.config/blender/2.77/scripts/addons/phobos/__init__.py", line 65, in <module>
>     import yaml
> ImportError: No module named 'yaml'

I think I have tracked the problem to the phobos __init__.py. Specifically line 65:

`sys.path.insert(0, sys.path[0] + "/phobos")`

Couple of issues with this. 1) Why are you blowing out the first element of the path? Shouldn't you be using append? 2) Regardless, on my machine sys.path[0] ($HOME/bin/blender-2.77a-linux-glibc211-x86_64/2.77/scripts/addons_contrib) does not correspond to where phobos is installed ($HOME/.config/blender/2.77/scripts/addons/phobos).

I believe line 65 should be changed to something like:

`sys.path.append(sys.path[-1] + "/phobos")`

And then line 67 of install_phobos.sh should change from:

`f.write("addonpath=$HOME/.config/blender/$blenderversion/scripts/addons")`

to:

`f.write("addonpath=$HOME/.config/blender/$blenderversion/scripts/addons/modules")`

But even this feels a little awkward... as relying on a specific entry in sys.path will probably get broken again in some future release of Blender.

kavonszadkowski, 2016-05-13:
Hi JD!

Thanks for the detailed report. We haven't encountered the error you describe before, but as you pointed out, some things tend to get broken by new releases of Blender once in a while. I'll see when I can give this a thorough check.

Meanwhile, you shouldn't have problems adding Phobos to your Blender addons manually, as the script is more a convenient way of installation (and handy when you're developing) rather than a necessity. Let me know if you run into any trouble with that.

BiYonic, 2018-02-23:
Hey JD,
we restructured the installation process of Phobos massively.
You can find the new information in the [Wiki](https://github.com/rock-simulation/phobos/wiki/Installation). Follow the steps there and everything should work properly.
I will close this issue for now. If you encounter any new issues, please open a new thread.
Happy blending,
Simon

closed 2018-02-23 by BiYonic