dfki-ric/phobos #108: ImportError: No mobule named 'phobos'

karlhm76, 2016-06-27:
Hello.

I am installing on a fresh install of Blender 2.77a
I have just downloaded the latest zip file for phobos and followed the installation instructions on the wiki.

I receive the error "No module named 'phobos'" when ticking the button to activate the addon.

[image]

I think the screenshot is attached.

Cheers,

Karl.

kavonszadkowski, 2016-06-27:
Hi Karl,

thanks for indicating this problem. We unfortunately do not have the resources to regularly test Phobos on all platforms with every new release of Blender.

This, though, looks like a purely Python-related problem, since there seems to be no "phobos" subfolder in your "scripts\addons" folder, thus Python cannot be expected to find that package. Guess something went wrong with the unzipping...?

karlhm76, 2016-06-27:
I'll retry unzipping. I was simply using the windows copy/paste on the zip file.

Thanks for the reply.

karlhm76, 2016-06-27:
Still no luck at all. 

I used winzip to unzip and rezip the zip file according to the instruction. I will try something else to generate my URDF.

Thanks for your help.

closed 2016-06-27 by karlhm76

kavonszadkowski, 2016-06-28:
I'm not sure about the details of zipping and unzipping on Windows in your case, but basically I was implying that the files in the zip folder have to end up in a "scripts\addons\phobos" subfolder, not directly in the "scripts\addons" folder.