dfki-ric/phobos #111: Problem activating Phobos in Blender

AmineCybernetics, 2017-02-02:
Hi all,

I followed the instruction in the tutorial to install phobos in blender. I am on ubuntu 14.04 and I am using blender 2.71. Wenn I tried to activate phobos in blender I get the error:
Traceback (most recent call last):
File "usr/share/blnder/2.71/scripts/addons/phobos/__init__.py", line 129, in <module>
 phobos.defs.updateDefs(os.path.dirname(__file__)+"/definitions")
AttributeError: 'module' object has no attribute 'defs'

I would be nice if you can help me fixing it :)

thanks in advance

kavonszadkowski, 2017-02-02:
I suspect you're working with the current master rather than the last release 0.7?
If so, please try the last release instead. There is a lot of work currently being done on master, see [Announcements](https://github.com/rock-simulation/phobos#announcements).

AmineCybernetics, 2017-02-02:
That solved the issue thanks!

kavonszadkowski, 2017-02-02:
Great to hear, hope you'll have a smoother run from here.

closed 2017-03-02 by kavonszadkowski

Leocorp, 2017-11-11:
hello, I just cloned the master (0.7) but I am still facing this issue "phobos has no attributes defs"
I'm working on Ubuntu 16.04, Blender 2.76b. Any help would be appreciated.