dfki-ric/phobos #115: URDF import is not supported

RDaneelOlivav, 2017-02-24:
I'm getting this error when importing an .urdf file

> [20170224_20:53] ERROR No import function available for selected model type: urdf (ImportModelOperator)
Traceback (most recent call last):
  File "/home/rdaneel/.config/blender/2.78/scripts/addons/phobos/operators/io.py", line 166, in execute
    models.buildModelFromDictionary(model)
UnboundLocalError: local variable 'model' referenced before assignment

Doesn't phobos have support for urdf files?
Have I installed something incorrectly?
Any Idea?

jacknlliu, 2017-02-28:
I got same problem.

BiYonic, 2017-02-28:
Hi,
Are you working with the **stable** branch or on the **master** branch?
Currently, there are a lot of changes on the master branch which might lead to your problem.
Greetings,
Simon

kavonszadkowski, 2017-03-01:
Thanks for the bug report. Simon is right, you should probably try stable, however, I'm not getting that error on master - which version of Blender are you guys using?

RDaneelOlivav, 2017-03-01:
I'm working on Blender 2.78, and the MasterBranch. Better in the stable then? Will it work in the Blender 2.78 version?

jacknlliu, 2017-03-01:
I'm using Blender 2.78, too.

kavonszadkowski, 2017-03-02:
That's odd, since I've been testing with Blender 2.78, too. The error apparently originates with Phobos not detecting the URDF import correctly (it's been plugin-based in Master for a while). I'll have a look into what could possibly go wrong there.

kavonszadkowski, 2017-03-02:
The last version that was tested with stable was 2.74. Blender sometimes breaks the API and/or the way the operators work, so that it's impossible to guarantee everything will work with 2.78.

kavonszadkowski, 2017-03-02:
Could you run Blender from terminal and paste the output on startup? There should be the following line if the URDF plugin loads correctly:

    Registering entity plugin: ['urdf']

BiYonic, 2017-03-29:
Hey @RDaneelOlivav and @jacknlliu,
do you have any update on the issue?
Was the error resolved in a way or does it still persist?
Greetings,
Simon

jacknlliu, 2017-03-30:
Sorry for arriving late to this thread.  Because the document of this project still not ready, it's still not easy-to -use for me. I hope to test this with the newer stable version.

jacknlliu, 2017-04-10:
I test this plugin (master branch) again with blender 2.78, but I still got error like following
```
$ blender

Read new prefs: /home/jack/.config/blender/2.78/config/userpref.blend
Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: /home/jack/.config/blender/2.78/scripts/addons/phobos/definitions
  defaultSensors.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultJoints.yml
  defaultSoftware.yml
  defaultMotors.yml
IMPORT:  phobos.io
Registering entity plugin: ['light']
Registering entity plugin: ['yaml']
Registering entity plugin: ['urdf']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['smurf']
Registering entity plugin: ['primitive']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobosgui
RELOAD:  phobos.phoboslog
IMPORT:  phobos.testing
RELOAD:  phobos.utils
Registering phobosgui...

measureit: Imported multifiles
Traceback (most recent call last):
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/operators/io.py", line 169, in execute
    models.buildModelFromDictionary(model)
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/models.py", line 849, in buildModelFromDictionary
    linkmodel.createLink(link)
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/links.py", line 92, in createLink
    inertiamodel.createInertialFromDictionary(link['name'], link['inertial'])
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/inertia.py", line 69, in createInertialFromDictionary
    assignMaterial(inert, 'phobos_inertial')
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/materials.py", line 97, in assignMaterial
    createPhobosMaterials()
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/materials.py", line 83, in createPhobosMaterials
    createMaterial(material, mat['diffuse'], mat['specular'], mat['alpha'], mat['diffuse_intensity'])
  File "/home/jack/.config/blender/2.78/scripts/addons/phobos/model/materials.py", line 58, in createMaterial
    mat.diffuse_color = diffuse
ValueError: bpy_struct: item.attr = val: sequences of dimension 0 should contain 3 items, not 9

location: <unknown location>:-1

location: <unknown location>:-1
```

Could this help?

BiYonic, 2017-04-10:
I just merged some bugfixes from the develop branch.
The problem: `ValueError: bpy_struct: item.attr = val: sequences (...)` should be resolved.
You can pull the changes and retry.
Maybe this fixes this issue, too. If so, please tell us and we will close it.
Greetings,
Simon

jacknlliu, 2017-04-10:
This plugin could not parse the urls of mesh in `.urdf` file which always begin with `package://` or `file:///`.
I got the following errors,
```
Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: /home/ros/.config/blender/2.76/scripts/addons/phobos/definitions
  defaultJoints.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultMotors.yml
  defaultSensors.yml
  defaultSoftware.yml
IMPORT:  phobos.io
Registering entity plugin: ['smurf']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['light']
Registering entity plugin: ['primitive']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
Registering entity plugin: ['urdf']
Registering entity plugin: ['yaml']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobosgui
RELOAD:  phobos.phoboslog
IMPORT:  phobos.testing
RELOAD:  phobos.utils
Registering phobosgui...
RNA_def_property_ui_text: 'outputMeshtype' '' description ends with a '.' !
RNA_def_property_ui_text: 'sceneName' '' description ends with a '.' !

RNA_def_property_ui_text: 'find' '' description ends with a '.' !
RNA_def_property_ui_text: 'replace' '' description ends with a '.' !
RNA_def_property_ui_text: 'add' '' description ends with a '.' !
[20170410_10:55] ERROR /data/catkin_ws/package:/ur_description/meshes/ur5/visual/shoulder.dae is no file. Object visual_0_shoulder_link will have empty mesh! (createGeometry)
[20170410_10:55] ERROR /data/catkin_ws/package:/ur_description/meshes/ur5/collision/shoulder.stl is no file. Object collision_0_shoulder_link will have empty mesh! (createGeometry)
Error: Cannot apply to a multi user: Object "collision_0_shoulder_link", Mesh "shoulder", aborting
Traceback (most recent call last):
  File "/home/ros/.config/blender/2.76/scripts/addons/phobos/operators/io.py", line 166, in execute
    models.buildModelFromDictionary(model)
  File "/home/ros/.config/blender/2.76/scripts/addons/phobos/model/models.py", line 849, in buildModelFromDictionary
    linkmodel.createLink(link)
  File "/home/ros/.config/blender/2.76/scripts/addons/phobos/model/links.py", line 110, in createLink
    bpy.ops.object.transform_apply(scale=True)
  File "/usr/share/blender/scripts/modules/bpy/ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Error: Cannot apply to a multi user: Object "collision_0_shoulder_link", Mesh "shoulder", aborting

location: <unknown location>:-1

location: <unknown location>:-1
```

kavonszadkowski, 2017-04-10:
The problem is that Phobos does not know about ROS paths and we don't usually work with our models as ROS packages. We fixed this a while ago, but apparently recently introduced a problem with the handling of relative file paths in case paths are instead given as ROS package paths. As even the old fix made a number of assumptions that may not be valid, this is something that should be thoroughly fixed anyway and I'll see what I can do (though I would like to add that I am not a big fan of depending on *rospack* for models that can be used by software outside of the ROS framework).

Amudtogal, 2018-08-10:
I currently see no point in checking for this error.
We are developing a software to open, edit and maintain different robot model formats.
There is no point in adding compatibility to any framework unless it refers to model formats.
This kind of problem can easily be avoided by moving meshes, urdf etc to a different folder and match the "Phobos" way of maintaining a model.
If anyone writes a small compatibility script, which moves the files back to ROS and updates the references in the URDF feel free to contact us and we might include it.

closed 2018-08-10 by Amudtogal