dfki-ric/phobos #116: Can not export URDF

LuKpL, 2017-05-05:
I'm getting this error when I try to export any robot as URDF

`Traceback (most recent call last):
File "/home/rfhmab308/.config/blender/2.76/scripts/addons/phobos/operators/io.py",line 129, in invoke
self.modelname = modellist[0]
File "/usr/share/blender/scripts/modules/bpy_types.py",line 600, in __setattr__
return setattr(properties,attr,value)
TypeError: bpy_struct: item.attr = val: expected a string enum, not tuple

location: <unknown location>:-1`

I'm using Blender 2.76 in Ubuntu. Does anyone have an idea?

kavonszadkowski, 2017-05-05:
Thanks for the info. There is always that one case you miss...just fixed it. Works for you now?

LuKpL, 2017-05-08:
I can export as URDF now but when I select .stl as mesh another error occurs and the urdf doesn't work

`Traceback (most recent call last):
File "/home/rfhmab308/.config/blender/2.76/scripts/addons/phobos/operators/io.py",line 130, in invoke
return self.execute(context)
File "/home/rfhmab308/.config/blender/2.76/scripts/addons/phobos/operators/io.py",line 147, in execute
exportModel(root, exportpath)
File "/home/rfhmab308/.config/blender/2.76/scripts/addons/phobos/operators/io.py",line 187, in exportModel
meshes.mesh_types[meshtype]['export'](model['meshes'][meshname],mesh_path)
File "/home/rfhmab308/.config/blender/2.76/scripts/addons/phobos/io/meshes//meshes.py",line 125, in exportMesh
bpy.ops.export_mesh.stl(filepath=outpath,use_selection=True, use_mesh_modifiers=True)
File "/usr/share/blender/scripts/modules/bpy/ops.py", line 189, in__call__ 
ret = op_call(sel.idname_py(),None,kw)
TypeError: Converting py args to operator properties: : keyword "use_selection" unrecognized

location: <unknown location>:-1`

kavonszadkowski, 2017-05-08:
Checked this error and found that it occurs due to an incompatibility in the Blender Python API between versions 2.78 (on which we are currently developing) and 2.76, which lacks the option 'use_selection' mentioned in the TypeError exception.
I'm gonna fix that to make it backwards compatible, but maybe you could consider changing to the latest Blender version in the long term. It's quite impossible to test our add-on with all Blender versions distributed with various flavors of Linux etc.

BiYonic, 2017-08-22:
This issue has been resolved with 1de28a75. Thus, I will close this. Feel free to reopen with objections.

closed 2017-08-22 by BiYonic