dfki-ric/phobos #118: Capsules

kavonszadkowski, 2017-08-30:
Currently, capsules for collision objects or visual objects are not supported by URDF and dealing with them is thus a bit problematic. However, there is [some discussion ](https://github.com/ros/urdfdom_headers/pull/24) to add them to URDF at some point, and they might at least end up in the successor of URDF.

Also, Blender allows to add them using its [Extra Objects add-on](https://blender.stackexchange.com/a/38578), which would have to be activated from within Phobos.

I propose to solve this issue by simply adding capsules as a geometry type and add an operator that allows to transform them into two cylinders and two spheres for URDF compatibility. Object properties such as simulation friction or collision settings etc. would have to be copied to all new objects.

Amudtogal, 2018-07-04:
Instead of replacing the capsules by other meshes, we could just calculate the output for sphere and cylinder on the fly, similar to the new Mesh Inertia Calculation (#146).

SDF does not handle capsules either. So we best, make this a util function.
We could even make this a generic tool, so users can replace more complex shapes (which are used alot) with their on configuration?

AlCap23, 2019-01-28:
Pushed to 1.1 . 

Could be done via Principle Component Analysis of the point cloud.