dfki-ric/phobos #121: Unit testing

kavonszadkowski, 2017-08-30:
We don't currently have any unit testing in Phobos. Unit testing is bit complicated, since Phobos is only run from within Blender. This could be solved by creating a test .blend file with a model including all possible components and a test script run from within that model that executes all functions.

Could make sense to put this test .blend in a different repository than Phobos itself.

Amudtogal, 2018-07-04:
I started adding assert statements at the beginning of our functions.
This helps us document the requested parameters and also might help us debugging (Instead of cryptic IndexErrors or KeyErrors) in the future, as the AssertExceptions are raised together with the provided description. See [here](https://www.tutorialspoint.com/python/assertions_in_python.htm) and for [this](https://wiki.python.org/moin/UsingAssertionsEffectively) for more detail.

Furthermore, after [building Blender from source](https://en.blender.org/index.php/User:Ideasman42/BlenderAsPyModule) we should be able to add testing suites to Blender.
Especially, we could follow the [CI approach](https://blender.stackexchange.com/questions/67274/how-to-do-continuous-integration-with-gitlab-when-developing-blender-addons#67275) with Travis which is described [here](https://anzui.de/en/blog/2015-05-21/) and file presets are [here](https://gitlab.com/d-e/blender-addon-boilerplate).

That way, we should get Phobos into a testable environment.

Amudtogal, 2019-01-28:
Dropping asserts again was a better decision.
We introduced model checking methods, that can be run as decorators, that allow to check different parameters of a visual, collisions etc
This does not enable proper unit tests, but we can introduce loads of test cases as soon as we prepare for the next release.

Amudtogal, 2019-08-15:
We have now Unit testing in Blender, see the latest commits 528b3129c41
Now we just need to add tests and keep on building our testing environment with blender files etc.

closed 2019-08-15 by Amudtogal