dfki-ric/phobos #123: Joint vs Motor limits

kavonszadkowski, 2017-08-30:
Currently joint and motor limits are used in parallel for URDF and SMURF. It needs to be settled if the mechanical limits of joints and the controller limits for motors in terms of range, and the physical limits in terms of speed or torque are different and should be handled thus.

Amudtogal, 2018-07-04:
In terms of Gazebo/SDF, there is only a joint limit, as motors are not described in their file format.

The question is: in which scenarios would it be required to use motor limits **and** joint limits?

Amudtogal, 2018-08-15:
I will delete motor limits when reintroducing motors to Phobos.

Amudtogal, 2018-09-03:
Since I added the motor overhaul, this should be resolved...

closed 2018-09-03 by Amudtogal