dfki-ric/phobos #126: YAML install problem with Blender 2.7x

Whistar, 2017-11-06:
I get following error when trying to activate phobos. I used the YAML included in the package. Tried it with Blender 2.72 and 2.74
Also tried under Windows 7 and Ubuntu 16.04
Apologies in advance as I'm sure I was the dunce in this case and misplaced something during install as both environments give similar errors but could find its proper root cause and solution:

on first try :

Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blende_2_71\2.71\scripts\modules\addon_utils.py", line 299, in enable
    mod = __import__(module_name)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 79, in import_submodules
    results.update(import_submodules(full_name))
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 75, in import_submodules
    results[full_name] = importlib.import_module(full_name)
  File "C:\Program Files\Blender Foundation\Blende_2_71\2.71\python\lib\importlib\__init__.py", line 104, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\yaml\cyaml.py", line 5, in <module>
    from _yaml import CParser, CEmitter
ImportError: No module named '_yaml'

Second click to activate:

Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blende_2_71\2.71\scripts\modules\addon_utils.py", line 299, in enable
    mod = __import__(module_name)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 79, in import_submodules
    results.update(import_submodules(full_name))
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\__init__.py", line 70, in import_submodules
    results[full_name] = importlib.reload(modules[full_name])
  File "C:\Program Files\Blender Foundation\Blende_2_71\2.71\python\lib\importlib\__init__.py", line 144, in reload
    methods.exec(module)
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\io\entities\__init__.py", line 8, in <module>
    py_mod = imp.load_source(mod_name, os.path.join(os.path.dirname(__file__), filename))
  File "C:\Program Files\Blender Foundation\Blende_2_71\2.71\python\lib\imp.py", line 169, in load_source
    module = methods.exec(sys.modules[name])
  File "C:\Users\PCC\AppData\Roaming\Blender Foundation\Blender\2.71\scripts\addons\phobos\io\entities\entities.py", line 30, in <module>
    import phobos.model.models as models
AttributeError: 'module' object has no attribute 'model'

ext/_yaml.c no such file or directory

What am I missing ?

Thanks in advance

BiYonic, 2017-11-09:
Hi Whistar,
at first, thanks for the bug report.
The problem with the shipped YAML modules has been around quite long now and we decided to remove them from Phobos in the current development phase.
Nevertheless, we decided to update the master branch with the improved installation script.
It will feature a python installation script which installs YAML by itself.
I will work on this hotfix and might finish it mid next week.
For the time being, are you sure, that you installed Phobos properly (as I read Blender 2.71 in the stack traces) for the right Blender version?
Best wishes

closed 2017-11-13 by BiYonic

BiYonic, 2017-11-13:
I hope this will fix the issue, please give some short feedback on the results.
Best wishes.

commit 79101e5b (2017-11-13): Hotfix: introduce python installation script

The installation shell script has been around for a while. However, it
does not help for Windows users. This problem is resolved by introducing
a python installation file instead.

Also, the problems with the included YAML installation are resolved by
removing it completely from phobos. However, this requires the user to
do a proper installation of YAML. For this reason, the installation page
in the wiki was updated to provide all required information.

The Windows part of the installation file is not tested yet.

Fixes #126.
commit 5e06bcd7 (2017-11-13): Hotfix: introduce python installation script

The installation shell script has been around for a while. However, it
does not help for Windows users. This problem is resolved by introducing
a python installation file instead.

Also, the problems with the included YAML installation are resolved by
removing it completely from phobos. However, this requires the user to
do a proper installation of YAML. For this reason, the installation page
in the wiki was updated to provide all required information.

The Windows part of the installation file is not tested yet.

Fixes #126.