dfki-ric/phobos #127: Unable to activate 'phobos' in Blender

Leocorp, 2017-11-11:
hello, I just cloned the master (0.7) but I am still facing this issue "phobos has no attributes defs"
I'm working on Ubuntu 16.04, Blender 2.76b. Any help would be appreciated.

**When blender opens, on first click to activate 'phobos', this is the error that pops in my console:**

Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: /home/eph/.config/blender/2.76/scripts/addons/phobos/definitions
  defaultSoftware.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultSensors.yml
  defaultJoints.yml
  defaultMotors.yml
IMPORT:  phobos.io
Registering entity plugin: ['yaml']
Registering entity plugin: ['primitive']
Registering entity plugin: ['urdf']
Registering entity plugin: ['light']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['smurf']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobos
Could not find yamlpath.conf
Using distributed package instead!
Importing phobos
RELOAD:  phobos.defs
Parsing definitions from: /home/eph/.config/blender/2.76/scripts/addons/phobos/definitions
  defaultSoftware.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultSensors.yml
  defaultJoints.yml
  defaultMotors.yml
RELOAD:  phobos.io
Registering entity plugin: ['yaml']
Registering entity plugin: ['primitive']
Registering entity plugin: ['urdf']
Registering entity plugin: ['light']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['smurf']
RELOAD:  phobos.model
RELOAD:  phobos.old_code
RELOAD:  phobos.operators
RELOAD:  phobos.phobos
Could not find yamlpath.conf
Using distributed package instead!
Importing phobos
RELOAD:  phobos.defs
Parsing definitions from: /home/eph/.config/blender/2.76/scripts/addons/phobos/definitions
  defaultSoftware.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultSensors.yml
  defaultJoints.yml
  defaultMotors.yml
RELOAD:  phobos.io
Registering entity plugin: ['yaml']
Registering entity plugin: ['primitive']
Registering entity plugin: ['urdf']
Registering entity plugin: ['light']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['smurf']
RELOAD:  phobos.model
RELOAD:  phobos.old_code
RELOAD:  phobos.operators
RELOAD:  phobos.phobos
Parsing definitions from: /home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/definitions
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 324, in enable
    mod = __import__(module_name)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/__init__.py", line 75, in import_submodules
    results[full_name] = importlib.import_module(full_name)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/__init__.py", line 70, in import_submodules
    results[full_name] = importlib.reload(modules[full_name])
  File "/usr/lib/python3.5/importlib/__init__.py", line 166, in reload
    _bootstrap._exec(spec, module)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/__init__.py", line 79, in import_submodules
    results.update(import_submodules(full_name))
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/__init__.py", line 75, in import_submodules
    results[full_name] = importlib.import_module(full_name)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/defs.py", line 180, in <module>
    updateDefs(os.path.dirname(__file__) + "/definitions")
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/phobos/defs.py", line 128, in updateDefs
    definitions['model']['sensors']['$forElem']['$selection__type'] = definitions['sensors']
KeyError: 'sensors'

**And then, when I try to re-activate a second time, I get this:**

Importing phobos
RELOAD:  phobos.defs
Parsing definitions from: /home/eph/.config/blender/2.76/scripts/addons/phobos/definitions
  defaultSoftware.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultSensors.yml
  defaultJoints.yml
  defaultMotors.yml
RELOAD:  phobos.io
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 324, in enable
    mod = __import__(module_name)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/__init__.py", line 138, in <module>
    import_submodules(phobos, verbose=True)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/__init__.py", line 79, in import_submodules
    results.update(import_submodules(full_name))
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/__init__.py", line 70, in import_submodules
    results[full_name] = importlib.reload(modules[full_name])
  File "/usr/lib/python3.5/importlib/__init__.py", line 166, in reload
    _bootstrap._exec(spec, module)
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/io/entities/__init__.py", line 8, in <module>
    py_mod = imp.load_source(mod_name, os.path.join(os.path.dirname(__file__), filename))
  File "/usr/lib/python3.5/imp.py", line 170, in load_source
    module = _exec(spec, sys.modules[name])
  File "/home/eph/.config/blender/2.76/scripts/addons/phobos/io/entities/yaml.py", line 4, in <module>
    import phobos.defs as defs
AttributeError: module 'phobos' has no attribute 'defs'

BiYonic, 2017-11-15:
Hi Leocorp,
the issue you are experiencing is related to the faulty YAML configuration.
This was fixed with 97dfd2427.
In order to update this, please pull the latest master version.
Then delete your installconfig.txt and yamlpath.conf from the phobos directory. (If you want to be really thorough also delete the phobos folder in your ~/.config/blender/2.76/scripts/addons).
This will reset your installation.
Afterwards run the new python installation file from console and enter the required data:
`./install_phobos.py`

Retry starting Blender and activating Phobos.
Please give some feedback, whether this did work or not.
Best wishes

Leocorp, 2017-11-15:
Hello BiYonic,
I just did everything you suggested (even deleted my phobos folder) and now I can't even get it to install.
This is what I get when I try running "python3 install_phobos.py"

python3 install_phobos.py 
install_phobos.py:101: SyntaxWarning: name 'phobospath' is assigned to before global declaration
  global phobospath
Enter your Blender version (e.g. "2.78") 2.76
What is your Python 3 command? (e.g. python3) python3
Phobos folder does not exist, create phobos folder in /home/eph/.config/blender/2.76/scripts/addons/phobos? (y/n) y
Traceback (most recent call last):
  File "install_phobos.py", line 127, in <module>
    if not installPhobos():
  File "install_phobos.py", line 110, in installPhobos
    return copyphobos(phobospath)
  File "install_phobos.py", line 84, in copyphobos
    copytree(path.join(os.getcwd(), 'phobos'), phobospath)
  File "install_phobos.py", line 49, in copytree
    shutil.copy2(s, d)
  File "/usr/lib/python3.5/shutil.py", line 251, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.5/shutil.py", line 115, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/eph/.config/blender/2.76/scripts/addons/phobos/phobosgui.py'

Any ideas why ??

BiYonic, 2017-11-15:
Hello Leocorp,
in which folder did you execute the install script?
It appears to me, that the script tried to copy from the addons folder (which is wrong). 
The FileNotFoundError is only caused by missing source data...
I just pushed another update, fixing the error you had previously (with the sensor definitions).
It was caused by a silent bug, which I could not see until removing Blender and Phobos and reinstalling anew.

Leocorp, 2017-11-15:
I am running the install script from the 'phobos' folder created after cloning with 'git clone'

BiYonic, 2017-11-15:
Same error as before?
Could you try using `./install_phobos.pz` instead of `python3` (the file should already be executable)?

Leocorp, 2017-11-15:
Tried that too. I think the error maybe originating from the 'copytree' function in the install_phobos.py.

BiYonic, 2017-11-15:
I cant reproduce this error. I completely removed Blender and all configs of it from the hard drive and downloaded it anew. And still, everything works fine (actually quite sad to be unhappy with not having an error at all).
What is your output of `git log`?
Could you again post the whole stacktrace?

Leocorp, 2017-11-15:
**this is my git log**
commit 3da00dcf0bf38080b813767eb4f8caae9076bf03
Author: Simon Reichel <simon.reichel@dfki.de>
Date:   Wed Nov 15 15:34:14 2017 +0100

    Hotfix misplaced Phobos definitions
    
    The activation of Phobos was failing because the definitions were placed
    in the wrong path. Also, the bl_info of the addon caused trouble due to
    a faulty string.
    
    Move the definitions folder to the appropriate place and fix the
    remaining installation issues.

commit 97dfd24273473f68f095d83495a2e45cc42af811
Author: Simon Reichel <simon.reichel@dfki.de>
Date:   Mon Nov 13 17:15:50 2017 +0100

    Hotfix Python installation
    
    The new installation led to some issues with yaml installations.
    Fix the copying of the yaml configuration and unify the configuration
    file endings.

commit 79101e5b20113145bcb69c214e432259c5e8a445
Author: Simon Reichel <simms1@web.de>
Date:   Tue Aug 22 13:58:13 2017 +0200

**Console output**
-MBP:~/Downloads/phobos$ ./install_phobos.py 
Found installation configuration.
Phobos folder does not exist, create phobos folder in /home/eph/.config/blender/2.76/scripts/addons/phobos? (y/n) y
Traceback (most recent call last):
  File "./install_phobos.py", line 128, in <module>
    if not installPhobos():
  File "./install_phobos.py", line 111, in installPhobos
    return copyphobos(phobospath)
  File "./install_phobos.py", line 85, in copyphobos
    copytree(path.join(os.getcwd(), 'phobos'), phobospath)
  File "./install_phobos.py", line 49, in copytree
    shutil.copy2(s, d)
  File "/usr/lib/python3.5/shutil.py", line 251, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.5/shutil.py", line 115, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/eph/.config/blender/2.76/scripts/addons/phobos/phobosgui.py'

Again I'm using Ubuntu 16.04 x64, Python 3.5, Blender 2.76

BiYonic, 2017-11-16:
I have no idea, where this FileNotFoundError comes from. It has to be a problem with your filesystem settings, as I don't experience this bug.
What happens if you copy the 'phobos' folder (which resides at the same level as the installscript) to the addons folder and run the script again?

Leocorp, 2017-11-16:
Tried that too, not working (same console output). Anyways thanks for your time. I think I'm gonna give up on trying to get phobos to run on my linux and install it on my mac rather.

BiYonic, 2017-11-16:
Yea, that's probably the best solution then.
Anyway, all you need to do is copy the 'phobos' folder into your addon directory, and create a file 'yamlpath.conf' which contains the absolute path to the folder where your yaml installation is saved in.
In Linux this is `/usr/local/lib/python3.5/dist-packages/yaml`. That way, Phobos should be able to start.

I will close this issue now. Good luck and happy blending!

closed 2017-11-16 by BiYonic

Leocorp, 2017-11-16:
Thanks man. The fix worked. Copied the phobos folder and created the 'yamlpath.conf'.
Off to 'Phobos Documentation'!!

Minaygt, 2024-06-30:
[image]
how can ı solve this error can you help me ?