dfki-ric/phobos #130: Installation on MacOS - Blender 2.77 [Solved Not Pretty]

Leocorp, 2017-12-11:
Using the normal "install_phobos.py" seem to be broken. Something I believe originates from the shutil.copy2() function.

After trying unsuccessfully to install the add-on as instructed, I finally got it to work (after tricking the install script) and here's how:
1- Copy the 'phobos' folder into your "../scripts/addons" folder.
2- In your install_phobos.py, change the if statement on line 127 to something like; 
     if False:
        .......
To get the script to skip trying to copy the folder for us.
3- Also change the string "yamlpath.conf" to something else. In my case "yamlpaths.conf" to cause the script to think you do not have the "yamlpath.conf" (which you don't) needs to be generated.
4- Save and run the "install_phobos.py" script.
5- Copy the generated "yamlpath.conf" to your phobos folder in step 1.

And everything should run smoothly. I think the issue should be investigated into further but for now, this is how it works.

NB: I have python3.6

Leocorp, 2017-12-11:
UPDATE ABOUT THE PROVIDED **startup.blend**

I have also found out the hard way that opening and saving the "startup.blend" provided with phobos as your default startup, kinda causes unexplained freezing of blender when next trying to work with "Phobos Operators" and even closing blender via the gui buttons.

Anyways I intend to keep the file around for when I need to import some materials from it (hoping it won't freeze blender over).

Another issue for devs.

neomanic, 2017-12-19:
@Leocorp, I've submitted this pull request to fix the installer under macOS: https://github.com/rock-simulation/phobos/pull/131 . The problem was an unneeded backslash in the directory path for the Blender addons. (This slash is needed on the shell command line, but not in quoted filenames in Python).

neomanic, 2017-12-19:
Worth noting, although I haven't looked into it too deeply, this appears to be a bug within the Python libraries. os.path.isdir(phobospath) (and .exists() too) returns True with that slash present, even though it definitely doesn't exist!

BiYonic, 2018-01-22:
Hey there,
we can close this now. The fixes have been introduced to both [release 0.7](/rock-simulation/phobos/tree/release-0.7) and the current [develop](/rock-simulation/phobos/tree/release-0.7) branches.
One problem fixed with the directory links, was the different path separators for different operating systems. This should work now, as expected.
Concerning the startup.blend we cannot help you unless you provide some more information on the issue. For example, startup blender from the commandline and paste it in a new issue.
Best wishes,
Simon

closed 2018-01-22 by BiYonic