dfki-ric/phobos #132: Installation on Windows

RokG, 2018-01-09:
For the life of me, I can't get it to work on Windows...

I followed the installation manual on two separate computers with blender 2.79 and both give the exact same error.
I downloaded the latest Python 3.6 (both 32 and 64 bit), then ran the 'pip3 install PyYaml' and finaly ran the 'python install_phobos.py' command. All without errors.
Apparently I have to use the 'python' and not 'python3' command on Windows? At least thats what I got from Googling (using python3 was my first problem I experienced). I got asked for Blender version and python command by the script anyway, to which I inputed 2.79 and python, respectively. The script didn't return any errors, but the addon was installed to some weird location??
C:\Users\Rok\AppData\Roaming\Blender\ Foundation\2.79\scripts\addons\phobos
Notice the space in the '\ Foundation' part. Also this is not where addons for Blender are on Windows.
I had to manually move the addon to the correct location, so that it even showed up in Blender user preferences.

When I try to enable the addon in Blender I get an error message saying: 'No module named 'yaml''.
Its on the line 107 of phobos/__init__.py file.

The yamlpath.conf has the path: C:\Users\Rok\AppData\Local\Programs\Python\Python36\lib\site-packages\yaml\ and folowing it gets me to a folder with some .py files.

I have close to zero experience with python and I have no clue what is going on?? Is there supposed to be a yaml.py file or something?

Any help is greatly appreciated!
Cheers

RokG, 2018-01-09:
Update:
Tried on a 3rd computer, running Windows7 (Previous ones were Windows 10)
Did the same thing as above, still the same problem.

Then I went to the Releases tab and downloaded the 7.0 version, which seems to work so far.
The 7.1 version was giving the same error.

The Installation guide was pointing to the https://github.com/rock-simulation/phobos and telling you to download the repository, shouldn't it be poinitng to **https://github.com/rock-simulation/phobos/*releases* instead?
Also, some comment said to download the stable branch, and not master branch. There is no stable branch.

RokG, 2018-01-09:
Update 2:
OK, this is wierd :) Tried to cleanup all extra files i created while trying to get this to work, and managed to break the addon again.
BUT! In the chaos of trying to get this work I figured this out:
I have to install both 0.7.1 and 0.7 version so that the 0.7.1 version works...

The 0.7.1 version has to be installed like I said in my first post, but the addon file then has to be moved to where the rests of the addons are (for me it is C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons). The yamlpath.conf also needs to be here.

The 0.7 version has to be installed like in the instructions (unzip the archive, then zip everything under phobos-0.7 folder to phobos.zip archive. Next, install the addon under User preferences->Addons->Install from file.

This will cause two Phobos in the Addons tab (one from 0.7 in the AppData location, and 0.7.1 one in ProgramFiles location). In order to be able to activate the addon, Activating the 0.7.1 phobos worked.

BiYonic, 2018-01-10:
Dear RokG,
I am on it. We hadn't tested the new installation on Windows (as we are working on Linux only machines).
Sorry for the whole mess. I will fix it and come back to you.
Greetings,
Simon

RokG, 2018-01-10:
Thanks for the feedback!

I tried importing a urdf file today, but it didn't work correctly. It only imported a few parts and some objects without the meshes, before I got a 'wrong location' error.
I think its a problem with the .urdf and not the addon, I'll have to look into it. Do you have any test urdf file that works 100%?

BiYonic, 2018-01-10:
Dear RokG,
I fixed the system paths in the installation script.
Typically, the user directory for blender addons is in MYHOME\AppData\Roaming\Blender Foundation\Blender\VERSIONNUMBER\scripts\addons. This is where the phobos folder should be after you ran the installation script.
The problem with the yaml package not being found is specific to Windows, as it does not distinguish between Python version 2 and 3. When you run `python` from commandline, it takes the version defined in your path variable which is supposedly not version 3.
I updated the documentation for this (using `py -3 install_phobos.py` should work), and also the python file, so it should invoke the right Python version in Windows anyway.
Sorry for the confusion with the stable branch. Some time ago, the master branch of Phobos was `develop` (which was far from stable). However, we switched these branches already, so if you clone the repository the master branch IS the stable branch. Development is found on the `develop` branch following this [git workflow](http://nvie.com/posts/a-successful-git-branching-model/).

The installation via zipfile is not working, as we have additional files and entities in a folder structure above the actual `phobos` addon (like the installation script).

The installation script should be fixed with the current hotfix.

It would be great if you could test the new installation file and check whether it works for you (both with `py -3` and `python`).

Good luck and greetings,
Simon

BiYonic, 2018-01-10:
To resolve this problem we need more details on the error you encountered.
Could you upload the stack trace when running from commandline?
We had no issues to this point with urdf files, so either we missed the issue you have (you wouldn't encounter it in our files), or there is an error in your file. Both ways an urdf file from us would not help...
I can only help you resolving the error cause.

RokG, 2018-01-10:
Hello BiYonic

Thanks for the update, I will try to reinstall later and get back to you.

In the meantime, this is what i get when i run blender from terminal:

```
Error: Cannot apply to a multi user: Object "collision_0_arm_right_3_link", Mesh "arm_3_collision", aborting
Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\operators\io.py", line 236, in execute
    models.buildModelFromDictionary(model)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\models.py", line 854, in buildModelFromDictionary
    linkmodel.createLink(link)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\links.py", line 110, in createLink
    bpy.ops.object.transform_apply(scale=True)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\modules\bpy\ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Error: Cannot apply to a multi user: Object "collision_0_arm_right_3_link", Mesh "arm_3_collision", aborting

location: <unknown location>:-1

location: <unknown location>:-1

```

Before I had some wrong paths in the .urdf file, I copied the robot from Gazebo on linux and i got this:

```
path not found
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\arm\arm_3.STL is no file. Object visual_0_arm_right_3_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\arm\arm_3_collision.STL is no file. Object collision_0_arm_right_3_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\gripper\inner_double.STL is no file. Object visual_0_gripper_left_inner_double_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\gripper\inner_double_collision.STL is no file. Object collision_0_gripper_left_inner_double_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\v2\knee_lo_res.stl is no file. Object visual_0_leg_right_4_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\v2\knee_collision.stl is no file. Object collision_0_leg_right_4_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\torso\torso_1.STL is no file. Object visual_0_torso_1_link will have empty mesh![2m (createGeometry)[0m
[20180110_16:18] [1m[91mERROR[0m D:\talos_description\meshes\torso\torso_1.STL is no file. Object collision_0_torso_1_link will have empty mesh![2m (createGeometry)[0m
Error: Cannot apply to a multi user: Object "collision_0_torso_1_link", Mesh "torso_1", aborting
Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\operators\io.py", line 236, in execute
    models.buildModelFromDictionary(model)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\models.py", line 854, in buildModelFromDictionary
    linkmodel.createLink(link)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\links.py", line 110, in createLink
    bpy.ops.object.transform_apply(scale=True)
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\modules\bpy\ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
RuntimeError: Error: Cannot apply to a multi user: Object "collision_0_torso_1_link", Mesh "torso_1", aborting

location: <unknown location>:-1

location: <unknown location>:-1
```

BiYonic, 2018-01-10:
This is a Blender error, not by Phobos. So I guess this is a problem with the URDF file we did not consider.
Could you send me a link for it (and the meshes)? Then I can try to debug the error.
Bw, Simon

kavonszadkowski, 2018-01-11:
Hi RokG,

the line causing the problem is supposed to apply the scale settings to newly created links to set them to a useful size in relation to the model. It can fail if the same meshes are used for both visual and collision objects, which we generally avoid due to the costly mesh collision calculation, which again is why this bug went undetected.

I just created a new version [0.7.3](/rock-simulation/phobos/releases/tag/0.7.3) that should fix this issue. We're currently in the process of changing our git workflow with the upcoming release, thus you can find this on the branch [release-0.7](/rock-simulation/phobos/tree/release-0.7).

You'll have to run:

    git fetch
    git checkout release-0.7

and install again.

RokG, 2018-01-11:
Hey,

Sorry for the late reply. I urgently had to do some renders, so I put this on hold. I'll try it on a separate PC now and let you both know.

I see, there are separate .stl files for the collision though.. however, maybe not all parts of the robots use them.
Could the error also be due to the fact that left and right arm of the robot use the same .stl files?

If I won't be able to get it work, I'll upload the .urdf file and all required files.

Cheers
Rok

kavonszadkowski, 2018-01-11:
I think it's only a problem within one link, but as I said, it should be fixed now.

RokG, 2018-01-11:
Well, the instalation now works OK.
I still had the problem with yaml, but I found the issue:

This is the path in yamlpath.conf: 
`C:\Users\Rok\AppData\Local\Programs\Python\Python36\lib\site-packages\yaml\`

Instead, it should be:
`C:\Users\Rok\AppData\Local\Programs\Python\Python36\lib\site-packages\`

I can now enable phobos and get the toolbar. Phobos stays activated after I exit and reopen Blender.

However, disabling and then enabling phobos again gives this error:
```
C:\Program Files\Blender Foundation\Blender>blender
Read prefs: C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\config\userpref.blend
found bundled python: C:\Program Files\Blender Foundation\Blender\2.79\python
Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos/definitions
  defaultJoints.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultMotors.yml
  defaultSensors.yml
  defaultSoftware.yml
IMPORT:  phobos.io
Registering entity plugin: ['generic']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['light']
Registering entity plugin: ['primitive']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
Registering entity plugin: ['urdf']
Registering entity plugin: ['yaml']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobosgui
RELOAD:  phobos.phoboslog
IMPORT:  phobos.testing
RELOAD:  phobos.utils
Registering phobosgui...

Unregistering Phobos...
Unregistering phobosgui...
Registering phobosgui...
Exception in module register(): 'C:\\Users\\Rok\\AppData\\Roaming\\Blender Foundation\\Blender\\2.79\\scripts\\addons\\phobos\\__init__.py'
Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender\2.79\scripts\modules\addon_utils.py", line 350, in enable
    mod.register()
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\__init__.py", line 145, in register
    phobos.phobosgui.register()
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\phobosgui.py", line 519, in register
    bpy.utils.register_class(PhobosPrefs)
RuntimeError: register_class(...):, missing bl_rna attribute from 'RNAMeta' instance (may not be registered)
```

After this, I'm unable to activate phobos again. But if exit blender and then open it again, then phobos is still activated, which is OK. So I don't know how urgent is this bugfix, just wanted to let you know.

RokG, 2018-01-11:
I can now confirm that it works with the urdf file here: https://github.com/openrobotics/talos_description

However, trying to import our custom urdf file, I get this error:

```
Read prefs: C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\config\userpref.blend
found bundled python: C:\Program Files\Blender Foundation\Blender\2.79\python
Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos/definitions
  defaultJoints.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultMotors.yml
  defaultSensors.yml
  defaultSoftware.yml
IMPORT:  phobos.io
Registering entity plugin: ['generic']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registering entity plugin: ['light']
Registering entity plugin: ['primitive']
Registering entity plugin: ['smurf']
Registering entity plugin: ['srdf']
Registering entity plugin: ['urdf']
Registering entity plugin: ['yaml']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobosgui
RELOAD:  phobos.phoboslog
IMPORT:  phobos.testing
RELOAD:  phobos.utils
Registering phobosgui...

Import finished in 0.0156 sec.
DAG zero... not allowed to happen!
DAG zero... not allowed to happen!
Import finished in 0.0312 sec.
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Import finished in 0.0000 sec.
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Import finished in 0.0000 sec.
Import finished in 0.0000 sec.
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Import finished in 0.0469 sec.
Import finished in 0.0000 sec.
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Import finished in 0.0312 sec.
Import finished in 0.0000 sec.
Import finished in 0.0312 sec.
Import finished in 0.0000 sec.
DAG zero... not allowed to happen!
DAG zero... not allowed to happen!
Import finished in 0.0000 sec.
Import finished in 0.0000 sec.
DAG zero... not allowed to happen!
Import finished in 0.0000 sec.
Import finished in 0.0156 sec.
Import finished in 0.0156 sec.
Import finished in 0.0156 sec.
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Import finished in 0.0000 sec.
DAG zero... not allowed to happen!
DAG zero... not allowed to happen!
Import finished in 0.0156 sec.
Import finished in 0.0000 sec.
Traceback (most recent call last):
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\geometries.py", line 143, in createGeometry
    if 'name' in viscol['material']:
KeyError: 'material'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\operators\io.py", line 236, in execute
    models.buildModelFromDictionary(model)
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\models.py", line 854, in buildModelFromDictionary
    linkmodel.createLink(link)
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\links.py", line 103, in createLink
    geometrymodel.createGeometry(visual, 'visual')
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\geometries.py", line 148, in createGeometry
    log('No material for obj', viscol['name'], 'DEBUG', 'createGeometry')
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\phoboslog.py", line 105, in log
    if loglevels.index(level) <= loglevels.index(prefs.loglevel):
ValueError: tuple.index(x): x not in tuple

location: <unknown location>:-1

location: <unknown location>:-1
```
However, compared to before, I now get at least some parts imported into Blender, so I guess the previous error is fixed.

This is the link to my urdf files and the blender file of what I get when I try to import. The urdf file is in the "robot" folder. I had to manulay change all paths from "package://talos_description/..." to my correct paths.
https://we.tl/f7ltsQCuR1

Cheers
Rok

kavonszadkowski, 2018-01-12:
Hi Rok,

thanks for the continuing reporting.

The registration bug is known, but we haven't looked into it, yet, as it doesn't stop you from using phobos. Sometimes Blender can be unwieldy with these registrations, and the documentation is incomplete with respect to a number of issues that can arise.

Fixed the second problem (was fixed on the develop branch before). You can pull the *release-0.7* branch.

RokG, 2018-01-12:
Hey Kai

Thank you all for your help.

Ok, I understand.

I pulled the release-0.7 branch. There is still the same problem with yamlpath.conf. The path inside points to the contents of the _yaml_ folder, instead to the _yaml_ folder itself. That is why blender reports an error that there is no model called _yaml_. 

The Import function is almost OK now. Both visual and collision meshes are imported (The model is not assembled though). All bones also seem to be added. Some bones also have constraints defined, but not all of them though (looks like it crashes during adding constraints?). The gripper of the robot is made from multiple meshes, but only one bone should control it (it just has one DOF), could that be the problem? I know how to make such mechanisms in Blender and it wouldn't take much time, so its no problem if this part is somehow skipped, so that the file finishes importing.

This is the error I get now:
```
Traceback (most recent call last):
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\operators\io.py", line 236, in execute
    models.buildModelFromDictionary(model)
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\models.py", line 859, in buildModelFromDictionary
    jointmodel.createJoint(joint)
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\joints.py", line 76, in createJoint
    setJointConstraints(linkobj, joint['type'], lower, upper)
  File "C:\Users\Rok\AppData\Roaming\Blender Foundation\Blender\2.79\scripts\addons\phobos\model\joints.py", line 222, in setJointConstraints
    for c in joint.pose.bones[0].constraints:
IndexError: bpy_prop_collection[index]: index 0 out of range, size 0

location: <unknown location>:-1

location: <unknown location>:-1
```

Cheers
Rok

BiYonic, 2018-01-17:
Hi Rok,
The confusion with the yaml path is dealt with in the newest commit on the release branch.
Actually, we stripped the yamlpath internally, which was unintuitive. Thus, your config file should be fine (as well as your yaml installation).
You can safely delete the yamlpath.conf, as it is replaced by python_dist_packages.conf, which contains the path to ALL of your additional python packages.
I will have a look at the other problem you are facing...
Simon

BiYonic, 2018-01-17:
Hi Rok,
The error you reported lies mostly in the urdf file. Some of the fixed joints (e.g. gripper_left_base_link_joint) involve the <axis> tag. Fixed joints in URDF do not have a rotation axis and this was causing the problem.
I added logging for this error in the current release. That way, the import will still succeed and the errors will be reported in the shell. This allows to debug the URDF file (at least concerning this specific problem).
On my side, it worked perfectly, although the geometry files are still not linked properly.
If it works for you, give us some feedback, so we can close this issue.
Good luck,
Simon

RokG, 2018-01-18:
Hey Simon,

I can confirm that it's working now (both instalation and importing urdf), you can close the issue. Thank you very much for your help!

For some reason, the grippers have the wrong mesh (instead of the gripper mesh there is a torso mesh).
It's just two meshes, so there won't be a problem replacing them manually. But otherwise, everything is at correct place and bones have correct limits. Unfortunately, I don't have any knowledge with .urdf files, but this is what we got with the robot.

By the way, is there any specific reason that each bone is in its own armature instead of just one armature with multiple bones?

Thanks again for all your help!
Cheers
Rok

BiYonic, 2018-01-18:
Dear Rok,
I am happy, that it worked out.
We chose to take single Blender armatures, because it is possible to parent them to each other in a way, that is easier to manage than extruding/adding single bones. Also, it is easier to assign custom properties to them and some more...
I know it seems counter intuitive, but it works like a charm.
Happy blending,
Simon

P.S.: a hint for the meshes, try debugging them using the "DEBUG" loglevel from the phobos preferences. Try to comment out sections of your urdf file to make it easier to grasp.

closed 2018-01-18 by BiYonic