dfki-ric/phobos #140: Texture export

hwiedDFKI, 2018-03-26:
Please allow to export textures with the path they were originally taken from.

kavonszadkowski, 2018-03-26:
Are we talking about absolute or relative paths?

hwiedDFKI, 2018-03-26:
Relative paths.
When exporting textures with phobos, it creates a new texture folder for the exported model. 
It would be nice if there would be an option to use the existing textures.

Amudtogal, 2018-07-04:
Why do we need this?

When using relative paths, the model will be portable and properly linked. By using absolute paths to reference an original texture on your drive, we lose this functionality.

What are the advantages to link existing textures with absolute paths?

hwied, 2018-07-04:
When having a model set/different versions of a model this would be useful to save memory. It's only a wish not a must have. 
As already set it shouldn't be absolute paths, but relative ones.
If you have a directory containing a model with textures and you create a version of this model somewhere in the same directory the textures can be reused and won't have to be exported again.

Amudtogal, 2018-07-04:
I understand the point.
To use the drive more efficiently we would have to solve a bunch of new problems:
- what if a texture has been updated externally (not from Blender), would we overwrite it?
- what if the texture has been deleted or moved from the external folder?
- what if the texture is in git supervision and the wrong branch is checked out?
- and so on...

From a programmer's point of view I would suggest, we try to keep external links (away from our Phobos defined folder structure) to a minimum.
This gives us the advantage, that we know where everything belongs. If a user wants to link a file on an external drive or similar, it is still possible to not export the textures from Phobos and link them manually in the respective files.

Another thought on this topic is, that Git integration of the model folder is awesome, as we can have the same model in different versions on different branches.

What may still be worth considering is the possibility for the user to insert arbitrary text as specific parameters which would then be passed on to the export. Is there a usecase for this?

Amudtogal, 2018-07-05:
We decided to not implement this feature for milestone 1.0.

closed 2018-07-05 by Amudtogal