dfki-ric/phobos #146: Mesh Inertia Calculation

AlCap23, 2018-06-05:
## Brief Description of the Issue

The inertia calculated via phobos/model/inertia/calculateMeshInertia , 340 ff. is not positive definite.

Furthermore the mesh calculation of the helper inertials is not available.

## Expected Behavior / New Workflow

Inertias shall be positive definite.

Helper Inertials of mesh geometries should be calculated.

## Actual Behavior / Current Workflow

See above.

I think the error responsible for the helper inertial calculation is located at line 159 in phobos/model/inertia. The object data is given as an argument, which does not contain the geometry data.

## Steps to Reproduce the Issue / Context in Which it Occurs

Calculate helper inertial from simple box defined as mesh.

## Operating System and Phobos Version

Ubuntu 16.04 / Blender 2.79b / Current Master Branch

kavonszadkowski, 2018-06-05:
Thanks for submitting the issue! This could have been two issues, but anyway, I just fixed the second problem, i.e. the exception thrown when calculating mesh inertia.

As for the problem with the implementation of mesh inertia calculation itself, we'll have to run a few more tests.

Amudtogal, 2018-06-28:
We could optimise the inertia calculation with numpy as @AlCap23 suggested.

However, let's get some measurements of the algorithm first, before optimisation!

Amudtogal, 2018-06-28:
We can also use the color of the misdefined inertias to debug the model in the checkModel operator...

Amudtogal, 2019-01-28:
We kept this issue just for reference of the possible coloring of bad inertias.

The inertia calculation was fixed some time ago, so we can consider this done.

closed 2019-01-28 by Amudtogal