dfki-ric/phobos #149: UI design Question (missing Button options)

boski04, 2018-06-20:
## Missing Button: Set Geometry Type
I am new to Phobos and trying to complete the steps of your Wiki.
Installation worked fine.
I am stuck at the step where you tell me to use the "Set Geometry Type" (https://github.com/dfki-ric/phobos/wiki/Modeling%20Walkthrough)
I tried all releases.
My Blender UI looks like this:
[image]
[image]
But on your installation guide it looks different.
Did you patch out this feature?
What am I supposed to do at this step?
Thank you very much. I hope this question is clear.
Korbi

kavonszadkowski, 2018-06-20:
Hi Korbi,
I am very sorry for the inconvenience. The button you are looking for is currently labelled *Define Geometry* and can be found in the *Visual/Collision* category.
We have been introducing a lot of changes in the code and the way the operators work recently, thus also adapting the GUI as we go. This is still far from reaching a stable state and so we haven't been keeping up with updating the documentation. We are currently working towards the next major release next month.
Kai

boski04, 2018-06-21:
Hi Kai,
thank you very much for your fast answer. I appreciate your work.
Looking forward to complete my first build.
Korbi

boski04, 2018-06-21:
Hi Kai,
I am still struggling with the whole building process, because the UI shown in the wiki differs to the UI of the actual release.
So my Question is: Is there a release matching the Wiki, that I can install? I tried a few, but couldnt find one.
Thanks for your attention.
Korbi

kavonszadkowski, 2018-06-21:
Hi Korbi,
that depends very much on the model you are building and the functionality you need for it, as it would  not help you if your UI looks like in the Wiki, but you cannot build the model you want to build. What is your project about, if I may ask?

Until we have finished our current codesprint and updated both the UI and the Wiki (and waiting for that may be an option as things are going to get cleaned up over the next weeks if you're not particularly on a schedule), it may be your best bet to not getting too worked up about the arrangement of the buttons or the exact wording of the labels. The overall process of building models as well as the related core functionality and labelling has not changed dramatically over the last releases.

Is there anything in particular you're still struggling with?

boski04, 2018-06-26:
Hi Kai,
I just need to complete a simple URDF of my 4-wheeled robot to present it in a gazebo simulation.
After that I will wait for your next release and then try to add more detail of sensors and exact movements to my robot.
So the main things I need right now are:
-Visuals and Links(already complete, i think)
-Colliders and Masses (partly done)
-Motors
Do you have some advice how to start. 
My problem is, you may have noticed that I'm not into robotics, but have good knowledge about Blender.(And my english is not the best)

Thank you very much.
Korbi

Amudtogal, 2018-06-27:
Hi Korbi,
as Kai has already pointed out, we need some more information to help you with your problem:

- where in the tutorial did you get stuck?
- have you checked the tooltips of the different Buttons?

Motors and sensors are not supported by URDF, so you do not need to spend time on that. You might need to configure them in Gazebo. In version 1.0 we will also support the import and export of SDF files which are native to Gazebo (see #142)...

Some advice to start is hard to give, if we do not know what your specific problems are... Therefore, try to follow the wiki guides as close as you can and tell us anywhere you get stuck (so we can help in a precise way). Sorry, it is not as clean as it should be! We are working on it :)

You might also get more information on your problem (if anything is not working as expected) from the debugging messages of Phobos. To see these, you need to run Blender from Terminal (or activating it in the "Help" Dropdown from within Blender) and set the log level in the Phobos User Preferences to 'DEBUG'. Error messages, warnings etc will then be printed to your commandline and you can check what went wrong. All these messages should be very verbose.

If you stumble upon a Python stacktrace or similar, please open an issue for that specific bug, so we can solve it.#

Good luck!
Simon