dfki-ric/phobos #151: Trouble Exporting URDF and Constructing Simple Robot

malceore, 2018-07-03:
Hello, I am using Blender 2.79, Phobos Release 7 and am having a hard time exporting a URDF robot made from scratch. I would just like to confirm that I am following the wiki correctly, I currently:
1. Import my meshes.
2. Select and set them as Visuals phobotype and Geometry types as mesh.
3. Create a link in Phobos menu and set it as root using the Phobos root button.
4. Select all meshes and bones then click robot and name robot.
5. Under export I have only URDF and DAE boxes checked when I attempt to export.

When I click export no files or errors are generated. Can anyone give me a simple todo list to make a very simple robot(maybe a four wheeled robot as an example) with only meshes and links in Phobos? Is having the collisions and inertia configured needed before a robot can be exported, I see when I change the logging to debug it seems to indicate this? Thank you, I would not bother the issues section if I had not already invested several days and scoured the wiki trying to resolve this.

Amudtogal, 2018-07-04:
Hi Brandon,
at first: welcome to Phobos!

I just followed your steps, creating a simple single-link robot with a standard cube as visual attached to it.
Following the steps you mentioned, I achieved this:
```
<?xml version="1.0"?>
<!-- created with Phobos 0.7 -->
  <robot name="test">

    <link name="Untitled">
      <visual name="Cube">
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <mesh filename="../meshes/dae/Cube.dae" scale="1.0 1.0 1.0"/>
        </geometry>
        <material name="Material"/>
      </visual>
    </link>

    <material name="Material">
      <color rgba="0.64 0.64 0.64 1.0"/>
    </material>

  </robot>
```
So your problem needs to be a different one. Could you check in the logs for errors and warnings (debug messages are only required if we were facing a bug)?
Also: have you provided an export path? Otherwise (if you have not saved a blendfile yet) the export will be placed relative compared to your Blender executable.

All in all, collisions and inertia need not to be specified, although a robot without them will hardly produce a great simulation. Thus, Phobos will produce a warning **only**, which may hint to misconfigured models. However, some things might be intended...

Happy blending,
Simon

malceore, 2018-07-04:
Thanks Simon, I am really hoping this will help me automate my robot simulations. With debug enabled and following recreating those steps I see this in the logs upon Export Model:
`
[20180704_12:41] INFO Parsing visual and collision (approximation) objects... (buildModelDictionary)
Traceback (most recent call last):
  File "/home/b/.config/blender/2.79/scripts/addons/phobos/operators/io.py", line 130, in invoke
    return self.execute(context)
  File "/home/b/.config/blender/2.79/scripts/addons/phobos/operators/io.py", line 147, in execute
    exportModel(root, exportpath)
  File "/home/b/.config/blender/2.79/scripts/addons/phobos/operators/io.py", line 153, in exportModel
    model = models.buildModelDictionary(root)
  File "/home/b/.config/blender/2.79/scripts/addons/phobos/model/models.py", line 759, in buildModelDictionary
    model['links'][parentname][obj.phobostype][nUtils.getObjectName(obj)] = props
KeyError: 'root'

location: <unknown location>:-1

location: <unknown location>:-1
`
Did I not setup Root correctly? One only has to click a link and then click Root under General Tools, right?

malceore, 2018-07-05:
I was able to import a simple URDF, modify it by adding two more cubes, two more links and getting a successful export. But then could not get my models attached rendered in RVIZ. I think I will give up for now and maybe hunt around at other pieces of URDF automation. Thanks again Simon!

Amudtogal, 2018-07-06:
Hey Brandon,
as the error log above suggests, this is a bug within Phobos we actually need to fix.
Would you mind checking out the master branch of Phobos and retrying?
I think we might have fixed this already, but it would help a lot to actually remove this bug.
Is one of your links named 'root' or similar? Or could you provide a blendfile for testing?

By the way, what exactly did not work in the end? Is it a problem with Phobos or Rviz?

malceore, 2018-07-06:
I could not get master nor release-0.8 to install using setup.py it tells me that Blender uses Python3.5 and not 3.6, but that is incorrect Blender lists 3.6 when following the wiki(going to scripts) and I do not have any other versions of Blender installed. I believe the exported file had slightly off coordinates when brought into RVIZ which I was able to use as a base to get my model work done. Is there anything else I can do to help troubleshoot?

Amudtogal, 2018-07-11:
Thanks for the info.
We will have a look into these problems for the new release of Phobos.
I don't know how RVIZ handles the mesh files, maybe the problem was on that side (meshes are exported without offset).

Good luck with your robot and we hope you will come back to Phobos, once 1.0 is out!

malceore, 2018-07-11:
I am eagerly awaiting a 1.0 release! If I get some time I will dig into the source and see if I can help as this project fully realized would be a great toolkit.

Amudtogal, 2018-08-15:
Thanks alot!
I will close this for now. Come back and reopen this, if the issue persists!

closed 2018-08-15 by Amudtogal